jneilliii/OctoPrint-YouTubeLive #4: Connection refused / Cannot assign requested address

dr3w3, 2018-01-29:
Hi,

thanks for your plugin - can't get it working.
Cloning works like a charm. 

While running the test, I get the following lines every time:

> pi@octopi:~ $ docker run --privileged --name YouTubeLive -ti octoprint/youtubelive:latest http://localhost:8080/?action=stream 1234-5678-90AB-CDEF
> OctoPrint Webcam Stream: http://localhost:8080/?action=stream
> Live-stream secret: 1234-5678-90AB-CDEF
> ffmpeg version N-86118-g6111ac7 Copyright (c) 2000-2017 the FFmpeg developers
>   built with gcc 4.9.2 (Raspbian 4.9.2-10)
>   configuration: --arch=armel --target-os=linux --enable-gpl --enable-omx --enable-omx-rpi --enable-nonfree
>   libavutil      55. 63.100 / 55. 63.100
>   libavcodec     57. 96.101 / 57. 96.101
>   libavformat    57. 72.101 / 57. 72.101
>   libavdevice    57.  7.100 / 57.  7.100
>   libavfilter     6. 90.100 /  6. 90.100
>   libswscale      4.  7.101 /  4.  7.101
>   libswresample   2.  8.100 /  2.  8.100
>   libpostproc    54.  6.100 / 54.  6.100
> [tcp @ 0x26fb560] Connection to tcp://localhost:8080 failed (Connection refused), trying next address
> http://localhost:8080/?action=stream: Cannot assign requested address

Searching the web tells me, that it may be an IPv4 <--> IPv6 problem, but I can't get it fixed.

Do you have some input for me?

Thank you

TheyCallMeZ, 2018-01-29:
I too am having this issue, I also tried changing localhost to the machine name and still got the same result.

jneilliii, 2018-01-29:
I was having this same issue on mine. I think it is related to how octopi's haproxy is configured. Can you try using the full url like below using your machines ip in the webcam settings and see if that works for you, it did for me. 

>http://<ip>:8080/?action=stream

TheyCallMeZ, 2018-01-29:
using the actual IP works for the docker container. I had updated the url in my octoprint config but the youtube live plugin still does not work.

dr3w3, 2018-01-29:
Hmm, that works for doing the test.
A few hours ago that also didn't work.

So with that the stream starts but If i want to start the stream via the plugin page in the web interface, I get that information:

YouTube Live Error
404 Client Error: Not Found ("No such container: abc123abc123abc123abc123abc123abc123)

....seems I need some help again? :-)

scns, 2018-01-29:
I've the same problem. see #2

scns, 2018-01-29:
Same issue  #5

jneilliii, 2018-01-29:
Can you guys verify that you have the latest version 0.3.0 of the plugin installed. I had noticed that I inadvertently left my YouTube Stream ID hardcoded in the plugin and as a result was getting other people's streams in my YouTube account. I reset my stream id, and could be part of the issue on your end.

TheyCallMeZ, 2018-01-29:
Yes I have 0.3.0 installed

scns, 2018-01-29:
Problem solved by setting the complete IP address in the OctoPrint Webcam stream URL:

http://LOCALIP/webcam/?action=stream

in stead off

/webcam/?action=stream

TheyCallMeZ, 2018-01-29:
> Problem solved by setting the complete IP address in the OctoPrint Webcam stream URL:
> http://LOCALIP/webcam/?action=stream
> in stead off
> /webcam/?action=stream

I did that but it seems I get stuck with a spinning go live button when i hit go live.

scns, 2018-01-29:
I've changed id, restarted the Pi and pressed play. This worked for me. Bud I had to restart the pi.

TheyCallMeZ, 2018-01-29:
Ok I'll give that a whirl after this print finishes. Thanks!

TheyCallMeZ, 2018-01-29:
that did resolve it, however I have a new problem! lol @jneilliii if you want me to open a ticket for this I can. 
short version is the youtube live doesn't do any camera rotation like what is set in the webcam options i'm using both 
Flip webcam horizontally
Flip webcam vertically

as my camera is mounted upside down.

scns, 2018-01-29:
See issue 6. I have the same issue

jneilliii, 2018-01-31:
New release applies rotation/flip options. Requires following commands to rebuild docker image.
- cd ~/youtubelive
- git pull
- docker build -t octoprint/youtubelive .

closed 2018-01-31 by jneilliii

Mutdogus, 2018-02-07:
I replaced localhost with my local ip address for my pi and I updated the full URL in the webcam settings.  Now I get "At least one output file must be specified".  See below for test output.

Filter:
ffmpeg version N-86118-g6111ac7 Copyright (c) 2000-2017 the FFmpeg developers
  built with gcc 4.9.2 (Raspbian 4.9.2-10)
  configuration: --arch=armel --target-os=linux --enable-gpl --enable-omx --enable-omx-rpi --enable-nonfree
  libavutil      55. 63.100 / 55. 63.100
  libavcodec     57. 96.101 / 57. 96.101
  libavformat    57. 72.101 / 57. 72.101
  libavdevice    57.  7.100 / 57.  7.100
  libavfilter     6. 90.100 /  6. 90.100
  libswscale      4.  7.101 /  4.  7.101
  libswresample   2.  8.100 /  2.  8.100
  libpostproc    54.  6.100 / 54.  6.100
Trailing options were found on the commandline.
Input #0, mjpeg, from 'http://10.0.1.29:8080/?action=stream':
  Duration: N/A, bitrate: N/A
    Stream #0:0: Video: mjpeg, yuvj420p(pc, bt470bg/unknown/unknown), 640x480, 5 tbr, 1200k tbn, 5 tbc
Guessed Channel Layout for Input Stream #1.0 : stereo
Input #1, s16le, from '/dev/zero':
  Duration: N/A, bitrate: 1411 kb/s
    Stream #1:0: Audio: pcm_s16le, 44100 Hz, stereo, s16, 1411 kb/s
At least one output file must be specified

jneilliii, 2018-02-07:
After making those settings changes for the webcam url did you restart octoprint?  Some have reported that is necessary.

jneilliii, 2018-02-07:
Also, I just realized I didn't update the docker instructions for testing after incorporating the flip/rotation options.  Use the following line to test the stream.

> docker run --privileged --name YouTubeLive -ti octoprint/youtubelive:latest http://localhost:8080/?action=stream xxxx-xxxx-xxxx-xxxx null

Mutdogus, 2018-02-08:
I did restart.  That didn't fix it though.  Adding the null at the end did.  However, I need to flip vertical, I'm assuming I need to replace "null" with a number like 180 or a "vertical" to get the feed flipped?

jneilliii, 2018-02-08:
Once you've successfully run the test command and it is sending the stream, the rest will take the settings directly from the webcam settings in OctoPrint for flip/rotate when you press the Go Live button on the YouTube Live tab.