jneilliii/OctoPrint-YouTubeLive #5: Unable to start stream

scns, 2018-01-29:
When I try the after the installation the test command everything works fine. 

(docker run --privileged --name YouTubeLive -ti octoprint/youtubelive:latest http://localhost:8080/?action=stream xxxx-xxxx-xxxx-xxxx) 

But when I press the play button in ocotprint nothing happens. There is no docker instance starting. 

When I press stop I'll receive the error: ouTube Live Error
404 Client Error: Not Found ("No such container: d541a9c3f6e358b570f8bc494e1c43aae2d13cdad1150c5003db4ac1aa215096")

scns, 2018-01-29:
Additional info: when i use debugging on Docker I see a "Stop request" but no start request:

Jan 29 19:13:12 octopi dockerd[5074]: time="2018-01-29T19:13:12.207051720+01:00" level=debug msg="Calling POST /v1.30/containers/5ea7c54169f173272a1b09d6520fc3ec3cf326515f17b5ba49883b6cc329660c/stop

scns, 2018-01-29:
after a reboot it seems to do something when I press the pay button.

`Jan 29 19:39:05 octopi dockerd[808]: time="2018-01-29T19:39:05.379777931+01:00" level=debug msg="Registering POST, /networks/{id:.*}/disconnect"
Jan 29 19:39:05 octopi dockerd[808]: time="2018-01-29T19:39:05.380285114+01:00" level=debug msg="Registering POST, /networks/prune"
Jan 29 19:39:05 octopi dockerd[808]: time="2018-01-29T19:39:05.380764068+01:00" level=debug msg="Registering DELETE, /networks/{id:.*}"
Jan 29 19:39:05 octopi dockerd[808]: time="2018-01-29T19:39:05.382621395+01:00" level=info msg="API listen on /var/run/docker.sock"
Jan 29 19:39:05 octopi ntpd[828]: Listen normally on 6 docker0 172.17.0.1 UDP 123
Jan 29 19:39:18 octopi dockerd[808]: time="2018-01-29T19:39:18.859572709+01:00" level=debug msg="Calling GET /v1.30/containers/YouTubeLive/json"

Jan 29 19:39:52 octopi dockerd[808]: time="2018-01-29T19:39:52.657108882+01:00" level=debug msg="Calling POST /v1.30/containers/create?name=YouTubeLive"
Jan 29 19:39:52 octopi dockerd[808]: time="2018-01-29T19:39:52.658244392+01:00" level=debug msg="form data: {\"AttachStderr\":false,\"AttachStdin\":false,\"AttachStdout\":false,\"Cmd\":[\"/webcam/?action=stream\",\".....\"],\"HostConfig\":{\"AutoRemove\":true,\"NetworkMode\":\"default\",\"Privileged\":true},\"Image\":\"octoprint/youtubelive:latest\",\"NetworkDisabled\":false,\"OpenStdin\":false,\"StdinOnce\":false,\"Tty\":false}"
Jan 29 19:39:52 octopi dockerd[808]: time="2018-01-29T19:39:52.910711907+01:00" level=debug msg="container mounted via layerStore: &{/var/lib/docker/overlay2/8764578bf4fbc772c9b442d0c5f30d4095d5e06f524fe8e23cbdfe34646e93a9/merged 0x1e6d4d0 0x1e6d4d0}"
Jan 29 19:39:52 octopi dockerd[808]: time="2018-01-29T19:39:52.912467307+01:00" level=debug msg="Probing all drivers for volume with name: 3c59ea834a8385f575d203b7545ac6751b517bc09997a41cecff61528474a72d"
Jan 29 19:39:52 octopi dockerd[808]: time="2018-01-29T19:39:52.914192500+01:00" level=debug msg="Registering new volume reference: driver \"local\", name \"3c59ea834a8385f575d203b7545ac6751b517bc09997a41cecff61528474a72d\""
Jan 29 19:39:52 octopi dockerd[808]: time="2018-01-29T19:39:52.937679945+01:00" level=debug msg="copying image data from 321277f4e41e9dcd41e6240e285181bfb465bf512dc859d610cf9096ef3c6752:/sys/fs/cgroup, to 3c59ea834a8385f575d203b7545ac6751b517bc09997a41cecff61528474a72d"
Jan 29 19:39:53 octopi dockerd[808]: time="2018-01-29T19:39:53.001410620+01:00" level=debug msg="Calling GET /v1.30/containers/321277f4e41e9dcd41e6240e285181bfb465bf512dc859d610cf9096ef3c6752/json"
Jan 29 19:39:53 octopi dockerd[808]: time="2018-01-29T19:39:53.085393560+01:00" level=debug msg="Calling POST /v1.30/containers/321277f4e41e9dcd41e6240e285181bfb465bf512dc859d610cf9096ef3c6752/start"
Jan 29 19:39:53 octopi dockerd[808]: time="2018-01-29T19:39:53.093886140+01:00" level=debug msg="container mounted via layerStore: &{/var/lib/docker/overlay2/8764578bf4fbc772c9b442d0c5f30d4095d5e06f524fe8e23cbdfe34646e93a9/merged 0x1e6d4d0 0x1e6d4d0}"
Jan 29 19:39:53 octopi dockerd[808]: time="2018-01-29T19:39:53.096021641+01:00" level=debug msg="Assigning addresses for endpoint YouTubeLive's interface on network bridge"
Jan 29 19:39:53 octopi dockerd[808]: time="2018-01-29T19:39:53.097260693+01:00" level=debug msg="RequestAddress(LocalDefault/172.17.0.0/16, <nil>, map[])"
Jan 29 19:39:53 octopi kernel: [   59.510328] docker0: port 1(veth5eb4deb) entered blocking state
Jan 29 19:39:53 octopi kernel: [   59.510335] docker0: port 1(veth5eb4deb) entered disabled state
Jan 29 19:39:53 octopi dockerd[808]: time="2018-01-29T19:39:53.126669542+01:00" level=debug msg="Assigning addresses for endpoint YouTubeLive's interface on network bridge"
Jan 29 19:39:53 octopi dockerd[808]: time="2018-01-29T19:39:53.153608257+01:00" level=debug msg="Programming external connectivity on endpoint YouTubeLive (7490f30e3f6a0f1a214b700221916bee2ba960532ff4815e6b64149541fc5ab1)"
Jan 29 19:39:53 octopi dockerd[808]: time="2018-01-29T19:39:53.159615963+01:00" level=debug msg="EnableService 321277f4e41e9dcd41e6240e285181bfb465bf512dc859d610cf9096ef3c6752 START"
Jan 29 19:39:53 octopi dockerd[808]: time="2018-01-29T19:39:53.159728410+01:00" level=debug msg="EnableService 321277f4e41e9dcd41e6240e285181bfb465bf512dc859d610cf9096ef3c6752 DONE"
Jan 29 19:39:53 octopi dockerd[808]: time="2018-01-29T19:39:53.187290505+01:00" level=debug msg="bundle dir created" bundle=/var/run/docker/containerd/321277f4e41e9dcd41e6240e285181bfb465bf512dc859d610cf9096ef3c6752 module=libcontainerd namespace=moby root=/var/lib/docker/overlay2/8764578bf4fbc772c9b442d0c5f30d4095d5e06f524fe8e23cbdfe34646e93a9/merged
Jan 29 19:39:53 octopi dockerd[808]: time="2018-01-29T19:39:53+01:00" level=debug msg="event published" module="containerd/containers" ns=moby topic="/containers/create" type=containerd.events.ContainerCreate
Jan 29 19:39:53 octopi dockerd[808]: time="2018-01-29T19:39:53+01:00" level=info msg="shim docker-containerd-shim started" address="/containerd-shim/moby/321277f4e41e9dcd41e6240e285181bfb465bf512dc859d610cf9096ef3c6752/shim.sock" debug=true module="containerd/tasks" pid=1144
Jan 29 19:39:53 octopi kernel: [   59.681752] docker0: port 1(veth5eb4deb) entered blocking state
Jan 29 19:39:53 octopi kernel: [   59.681756] docker0: port 1(veth5eb4deb) entered forwarding state
Jan 29 19:39:53 octopi kernel: [   59.681839] IPv6: ADDRCONF(NETDEV_CHANGE): docker0: link becomes ready
[... cut ...]
Jan 29 19:39:55 octopi dockerd[808]: time="2018-01-29T19:39:55.535122035+01:00" level=debug msg="Removing volume reference: driver local, name 3c59ea834a8385f575d203b7545ac6751b517bc09997a41cecff61528474a72d"
Jan 29 19:39:56 octopi dhcpcd[803]: docker0: carrier lost
Jan 29 19:39:56 octopi dhcpcd[803]: docker0: deleting address fe80::ae65:bd8a:589c:5bb8
Jan 29 19:39:56 octopi avahi-daemon[454]: Withdrawing address record for fe80::ae65:bd8a:589c:5bb8 on docker0.
Jan 29 19:39:56 octopi avahi-daemon[454]: Leaving mDNS multicast group on interface docker0.IPv6 with address fe80::ae65:bd8a:589c:5bb8.
Jan 29 19:39:56 octopi avahi-daemon[454]: Interface docker0.IPv6 no longer relevant for mDNS.
`

dr3w3, 2018-01-29:
Yeah, that's the output I've logged a few days ago at syslog, because of that I thought of an IPv6 <---> IPv4 problem...

jneilliii, 2018-01-29:
Can you guys verify that you have the latest version 0.3.0 of the plugin installed.  I had noticed that I inadvertently left my YouTube Stream ID hardcoded in the plugin and as a result was getting other people's streams in my YouTube account.  I reset my stream id, and could be part of the issue on your end.

scns, 2018-01-29:
Problem solved by setting the complete IP address in the OctoPrint Webcam stream URL:

http://LOCALIP/webcam/?action=stream 

in stead off

/webcam/?action=stream

closed 2018-01-29 by scns