jneilliii/OctoPrint-YouTubeLive #6: Flip stream

scns, 2018-01-29:
In OctoPrint it's possible to flip the webcam feed.

[image]

These settings are not used in the Youtube Feed. How can I transfer these settings into the feed?

jneilliii, 2018-01-29:
Yeah, I figured this request would be on its way when I saw the streams posted to my account being upside down. It may take me some time to figure that out as I'm not super familiar with ffmjpeg, but essentially you need to modify the entry.sh file that is cloned from github for the proper parameters to the ffmjpeg command.

scns, 2018-01-31:
> New release 0.4.0 should work for this. Requires rebuilding of docker
> image with updated entry.sh file.

Nice. Thanks for the great work!

Works fine now!

Met vriendelijke groet,

*Maarten Schmeitz*