jneilliii/OctoPrint-YouTubeLive #9: 404 Client Error:

pythonjohn, 2018-02-08:
YouTube Live Error
404 Client Error: Not Found ("No such container: a575b9ee935d07602f13dfd467647347a8751e

Not sure what happening but Installed using copy and paste from the setup page. 
Using Raspberry Pi 3

jneilliii, 2018-02-08:
Try changing your webcam stream url to one that includes the ip address of your pi.  It would be something like below:

> http://192.168.1.110:8080/?action=stream

Bigwerth, 2018-02-24:
Unable to find image 'octoprint/youtubelive:latest' locally
docker: Error response from daemon: pull access denied for octoprint/youtubelive, repository does not exist or may require 'docker login'.
See 'docker run --help'.

WheresWaldo, 2018-03-04:
I will verify that this is an issue. Streaming direct from the IP address and you get the 'No such container message" when you stop the stream. On rPi3 with raspicam.

jneilliii, 2018-03-04:
Yes, this plugin is not designed to stream direct from raspicam. It is designed to reencode the mjpg stream being created by octoprint. By doing it this way both this plugin and the console tab will be able to live together. If you want to do direct stream you have to kill the webcamd service, change the command line in the github file that was cloned and you lose the ability to see the stream on the console tab.

jneilliii, 2018-03-04:
@WheresWaldo, if you want to do that specifically you would edit the entry.sh downloaded from the github prior to building the docker image to be something like this...

> raspivid -o - -t 0 -w 1920 -h 1080 -fps 40 -b 8000000 -g 40 | ffmpeg -re -ar 44100 -ac 2 -acodec pcm_s16le -f s16le -ac 2 -i /dev/zero -f h264 -i pipe:0 -c:v copy -c:a aac -ab 128k -g 40 -strict experimental -f flv -filter:v $3 -r 30 rtmp://a.rtmp.youtube.com/live2/$2

This is a command line I grabbed from the original [source](https://gist.github.com/alexellis/b86a91225eabd004573fe09da3fb34b2), but like previously mentioned though, you have to stop the webcamd service, etc. This would require a good amount of testing.

WheresWaldo, 2018-03-04:
@jneilliii maybe this should be a feature request and not piggybacked on @pythonjohn's issue.

I am specifically trying to get this to work on the rPi camera module since I have a Robo R2 that has this camera preinstalled on it's built in rPi3. And the normal avconv streaming is problematic on YouTube Live.

Namain, 2018-03-27:
I'm also having this issue. I successfully tested streaming using the docker run command but I'm not able to stream via the Octoprint website on my Raspberry Pi 3. I tried setting the webcam stream URL to http://192.168.32.21:8080/webcam/?action=stream (my Pi's IP), but the test fails when I do that. If I set the stream URL to http://127.0.0.1:8080/webcam/?action=stream the test works, but it the stream still fails.

I'm also using a Logitech C615 camera like @pythonjohn

jneilliii, 2018-03-27:
Did you restart the octoprint service after settings the steam url?

Namain, 2018-03-28:
I did but looks like this is my fault. I was using an ssh tunnel (similar
to a VPN, but works slightly differently) to setup the octoprint while I
was at work and the tunnel was messing with the streams

closed 2018-08-05 by stale[bot]

totalitarian, 2019-07-24:
Same issue here using usb webcams