jneilliii/OctoPrint-YouTubeLive #11: YouTube Live Error

trev502000, 2018-02-20:
i constantly get this error is this a bug or me being stupid?

YouTube Live Error ('Connection aborted.', error(2, 'No such file or directory'))

YouTube ids are all correct and camera is a logitech c270

jneilliii, 2018-02-20:
Did you set the webcam url for the stream in octoprint to be a fully qualified url using the ip address?

trev502000, 2018-02-21:
yes i have done that and no change

jneilliii, 2018-02-21:
And you followed the instructions for getting everything set up and configured?

jneilliii, 2018-02-21:
Does the test command in the instructions work for you?

trev502000, 2018-02-23:
it works perfect now started from scratch i must have done something wrong 
thank you for your help its a brilliant add on

closed 2018-02-23 by trev502000

john5634, 2018-03-17:
I get the error referenced above and I fixed my url to look like this: http://10.0.0.49/webcam/?action=stream. And I still get the "YouTube Live Error
('Connection aborted.', error(2, 'No such file or directory'))" error, and I am 100% sure I have the channel ID and stream id correct.

Any ideas?

jneilliii, 2018-03-17:
After changing the stream url I believe you need to restart octoprint. Sorry, it's been a while, but I think I remember some others reporting that.

john5634, 2018-03-17:
No luck, just restarted it and rebooted the whole pi as well. Any help would be greatly appreciated! I can even give you a temp login to my octopi if you want!

jneilliii, 2018-03-18:
Did you follow [these](https://github.com/jneilliii/OctoPrint-YouTubeLive/blob/master/docker_instructions.md) instructions?

DarasuumNebelin, 2018-08-08:
Were do I input these commands exactly ?

jneilliii, 2018-08-08:
You have to ssh into your pi or connect a keyboard/monitor and log into the os.

DarasuumNebelin, 2018-08-08:
Did everything as described but had some trouble:
the test didn't work said there is no directory
and in octoprint i when i pushed "go live" I got that: 

404 Client Error: Not Found ("No such container: 737c53f5776968a7c84b460caba7d4562b59bf3b0f14cf8d9ecb9c8a00613afa")

DarasuumNebelin, 2018-08-08:
Hi Thanks for the help!
You were right I put the IP address in wrong. 
Got the test working (and typed "docker rm YouTubeLive")

But on the Octoprint side I still get the same message.

404 Client Error: Not Found ("No such container: 737c53f5776968a7c84b460caba7d4562b59bf3b0f14cf8d9ecb9c8a00613afa")

Thou if I ssh into my octoprint and start a stream it works even in octoprint but I can't close it or open it.
Any Ideas ?
I mean it is working, but for convenience sake the one button method of the plugin would be nice.
Thank you in advance!!

DarasuumNebelin, 2018-08-09:
Got it working! Thx
But the resolution is still bad!
I have 640x480, thou the default should be 1080x1080
Any idear ?

jneilliii, 2018-08-09:
The resolution was set low to try and reduce the lag. I may update the plugin in the future to allow the user to adjust that. I'm not 100% sure what mjpgstreamer is set for so there may be a ceiling on how high it can go anyway since the plugin just re-encodes that stream.

DarasuumNebelin, 2018-08-09:
OK, thank you anyway for the help. Much appreciated !!