jneilliii/OctoPrint-YouTubeLive #12: Unable to find image 'octoprint/youtubelive:latest' locally

Bigwerth, 2018-02-24:
Unable to find image 'octoprint/youtubelive:latest' locally
docker: Error response from daemon: pull access denied for octoprint/youtubelive, repository does not exist or may require 'docker login'.
See 'docker run --help'.

Don't know the docker user name and password, wasn't asked in the installation.

jneilliii, 2018-02-24:
You missed a few steps in the instructions. That docker image is built locally after cloning a git.

closed 2018-02-24 by jneilliii

Bigwerth, 2018-02-24:
I followed the instruction:
Install docker

curl -sSL https://get.docker.com | sh
sudo usermod pi -aG docker
sudo reboot

Pull Docker Image

docker pull alexellis2/streaming:17-5-2017

Clone Repository and Rebuild

cd ~
git clone https://github.com/jneilliii/youtubelive --depth 1
cd youtubelive
docker build -t octoprint/youtubelive

What do I miss?

jneilliii, 2018-02-25:
In this part

cd youtubelive
docker build -t octoprint/youtubelive

Make sure you run the sucker build command with a space followed by a period. Like this...

cd youtubelive
docker build -t octoprint/youtubelive .

jneilliii, 2018-02-25:
Once you build you can run the test command I  the instructions and verify it's working.