jneilliii/OctoPrint-YouTubeLive #15: Test run of docker fails

bipsendk, 2018-03-11:
When trying a test run of docker (according to instructions), I get:

OctoPrint Webcam Stream: http://192.168.225.235:8080/?action=stream
Live-stream secret: xxxxxxxxxxxxxxxxxxxxxx
Filter: null
ffmpeg version N-86118-g6111ac7 Copyright (c) 2000-2017 the FFmpeg developers
  built with gcc 4.9.2 (Raspbian 4.9.2-10)
  configuration: --arch=armel --target-os=linux --enable-gpl --enable-omx --enable-omx-rpi --enable-nonfree
  libavutil      55. 63.100 / 55. 63.100
  libavcodec     57. 96.101 / 57. 96.101
  libavformat    57. 72.101 / 57. 72.101
  libavdevice    57.  7.100 / 57.  7.100
  libavfilter     6. 90.100 /  6. 90.100
  libswscale      4.  7.101 /  4.  7.101
  libswresample   2.  8.100 /  2.  8.100
  libpostproc    54.  6.100 / 54.  6.100
[mjpeg @ 0x2fcb130] unable to decode APP fields: Invalid data found when processing input
Input #0, mjpeg, from 'http://192.168.225.235:8080/?action=stream':
  Duration: N/A, bitrate: N/A
    Stream #0:0: Video: mjpeg, yuvj422p(pc, bt470bg/unknown/unknown), 1280x720, 5 tbr, 1200k tbn, 5 tbc
Guessed Channel Layout for Input Stream #1.0 : stereo
Input #1, s16le, from '/dev/zero':
  Duration: N/A, bitrate: 1411 kb/s
    Stream #1:0: Audio: pcm_s16le, 44100 Hz, stereo, s16, 1411 kb/s
Stream mapping:
  Stream #0:0 -> #0:0 (mjpeg (native) -> h264 (h264_omx))
  Stream #1:0 -> #0:1 (pcm_s16le (native) -> aac (native))
Press [q] to stop, [?] for help
[mjpeg @ 0x2fcb7f0] unable to decode APP fields: Invalid data found when processing input
[swscaler @ 0x305b950] deprecated pixel format used, make sure you did set range correctly
[h264_omx @ 0x300d640] Using OMX.broadcom.video_encode
[h264_omx @ 0x300d640] OMX error 80001000
[h264_omx @ 0x300d640] err 80001018 (-2147479528) on line 561
Error initializing output stream 0:0 -- Error while opening encoder for output stream #0:0 - maybe incorrect parameters such as bit_rate, rate, width or height
Conversion failed!

The URL for my local video stream works without problems....

jneilliii, 2018-03-11:
Did you set privileges and reboot as part of the docker install process? 

sudo usermod pi -aG docker
sudo reboot

nowkillkennys, 2018-07-04:
im getting this aswell

jneilliii, 2018-07-04:
Which version of octopi are you running @nowkillkennys? I made an update recently that changed the docker pull from the original source.

jneilliii, 2018-07-04:
Also check out #31.

closed 2018-08-05 by stale[bot]