jneilliii/OctoPrint-YouTubeLive #16: Unable to decode APP Fields.

Whyhacks, 2018-03-16:
Hey there.

When using: docker run --privileged --name YouTubeLive -ti octoprint/youtubelive:latest http://localhost:8080/?action=stream xxxx-xxxx-xxxx-xxxx null ( With filled in xxxx and so on ) 

the livestream starts. But right now it did run around 5 minutes.
SSH tells me unable to decode APP Fields. Invalid data found when processing input.

So what can i do ?

jneilliii, 2018-03-16:
Lots of Google searching. I don't have much knowledge of any of the individual pieces that go together to make this work, just the plugin side to initiate the individual pieces.

Whyhacks, 2018-03-16:
Yeah but it doesnt seem like the plugin side does what it should do ?

I sticked to the tutorial and tried it 4 times now. 
The only thing installed on the raspberry is Octroprint.

jneilliii, 2018-03-16:
So your octoprint is not from an octopi image?

Whyhacks, 2018-03-16:
Ok. Then thanks for your good help and your perfect working plugin.

Whyhacks, 2018-03-17:
Got it working. 
But only for 5 minutes ^^.
Then the PI crashed.

Tried to reinstall it and same failure again.

I downloaded octoprint from the official site.
Installed / Set Wlan Things / Expanded File System / Updatet / tried to install the Plugin.

Dont know if iam doing something wrong.

Or does the last point of the tutorial just takes like ever ?

Whyhacks, 2018-03-17:
Again got it up and running.

Had to change Octoprint / Settings / Webcam : 

webcam/?action=stream

to

http://local IP/webcam/?action=stream

Now it starts and stop´s like it should. Delay is heavy but thats ok for me.

jneilliii, 2018-03-17:
Yeah, that was documented in the [instructions](https://github.com/jneilliii/OctoPrint-YouTubeLive/blob/master/docker_instructions.md), glad you got it working. I agree, the delay is much longer than I'd like, usually 30-45 seconds for me, but I think that's not to bad taking into consideration its having to reencode the mjpg stream. I think it would be more efficient for sure if it removed the mjpg stream and we were using the camera as input directly.

closed 2018-03-17 by jneilliii