jneilliii/OctoPrint-YouTubeLive #19: no tab and config after enable from 1.37rc4

fred2088, 2018-04-05:
no tab and config page after enable from plugin manager on 1.37rc4

jneilliii, 2018-04-05:
Thanks, I'll test that on my end and see what's going on.

jneilliii, 2018-04-05:
I've tested on my RC4 installation and am not having any issues as you described.  I uninstalled, deleted the previous settings and installed from the plugin manager as if it was a brand new installation.

Are there any errors in your octoprint.log or browser's debug console?

fred2088, 2018-04-06:
tried on Pi3 and Pi zero work fine, but not from Windows while it install successfully ...

jneilliii, 2018-04-06:
That makes sense, I only use Pis.  Based on the error though, it looks like you may be able to install pypiwin32 maybe?

fred2088, 2018-04-07:
still can't get it work after installed pypiwin32 ?

fred2088, 2018-04-08:
yes

closed 2018-08-05 by stale[bot]