jneilliii/OctoPrint-YouTubeLive #22: ('Connection aborted.', error(13, 'Permission denied'))

REVIEWGAMINGMAN, 2018-05-08:
('Connection aborted.', error(13, 'Permission denied'))

jneilliii, 2018-05-08:
Did you follow the instructions posted [here](https://github.com/jneilliii/OctoPrint-YouTubeLive/blob/master/docker_instructions.md)? At what point do you get this error?

REVIEWGAMINGMAN, 2018-05-08:
yah

REVIEWGAMINGMAN, 2018-05-08:
I have purged everything on docker

REVIEWGAMINGMAN, 2018-05-08:
Got permission denied while trying to connect to the Docker daemon socket at unix:///var/run/docker.sock: Post http://%2Fvar%2Frun%2Fdocker.sock/v1.37/containers/prune: dial unix /var/run/docker.sock: connect: permission denied

REVIEWGAMINGMAN, 2018-05-08:
https://www.youtube.com/watch?v=zWq0wAPEfsk

REVIEWGAMINGMAN, 2018-05-08:
OctoPi 
yah 1.5

REVIEWGAMINGMAN, 2018-05-08:
container related

REVIEWGAMINGMAN, 2018-05-08:
pi@octopi:~/youtubelive $ Error: No such container: YouTubeLive

REVIEWGAMINGMAN, 2018-05-08:
root@octopi:~/youtubelive# docker build -t octoprint/youtubelive .
Sending build context to Docker daemon  53.76kB
Step 1/3 : FROM alexellis2/streaming:17-5-2017
 ---> 63e724cc90ea
Step 2/3 : COPY entry.sh entry.sh
 ---> 67c33824c572
Step 3/3 : RUN chmod +x entry.sh
 ---> Running in aaa1c77df964
Removing intermediate container aaa1c77df964
 ---> fe69f8b40b6d
Successfully built fe69f8b40b6d
Successfully tagged octoprint/youtubelive:latest
root@octopi:~/youtubelive# docker rm YouTubeLive
Error: No such container: YouTubeLive
root@octopi:~/youtubelive#

jneilliii, 2018-05-09:
You missed the testing line in the instructions where you run the following command replacing localhost with your pi's ip address and `xxxx-xxxx-xxxx-xxxx` with your YouTube Live Stream ID.

> docker run --privileged --name YouTubeLive -ti octoprint/youtubelive:latest http://localhost:8080/?action=stream xxxx-xxxx-xxxx-xxxx null

Once you run that command it should return a bunch off ffmpeg output to the terminal/console. Using your browser check to verify the stream is running from your [YouTube Live Dashboard](https://www.youtube.com/live_dashboard). Then you'll need to stop and remove the docker image by holding down the `CTRL` key on your keyboard followed by the letter `C`. Then type the following command.

> docker rm YouTubeLive

closed 2018-08-05 by stale[bot]