jneilliii/OctoPrint-YouTubeLive #27: "Go Live" doesnt work after sucessful test in the shell.

Pitta, 2018-06-10:
The stream player loads with my account, but the go live button greys out and animates like its working after I click it. It never starts the stream. When I refresh, I get the stop button which spite out the following truncated error message:

```
404 Client Error: Not Found ("No such container: 2e5c2adcde7bfeeba045826829a68cc46653
```

jneilliii, 2018-06-10:
Did you follow the instructions [here](https://github.com/jneilliii/OctoPrint-YouTubeLive/blob/master/docker_instructions.md)?

jneilliii, 2018-06-10:
The last thing to check would be if your webcam stream url is set with the ip address and you restarted octoprint?

jneilliii, 2018-06-10:
Thanks for pointing that out.  That's the default stream url for mjpg streamer.  Try using something like this `http://192.168.1.101/webcam/?action=stream`.

Pitta, 2018-06-10:
OK. I've done that. Test in webcam settings pulls up feed, but it's upside down.

YouTube live "Go Live" button still gets stuck spinning.

Pitta, 2018-06-10:
The logs for youtubelive just say:
```
2018-06-10 03:00:20,633 - octoprint.plugins.youtubelive - INFO - Checking stream status.
2018-06-10 03:00:24,880 - octoprint.plugins.youtubelive - INFO - Stop stream command received.
2018-06-10 03:00:29,505 - octoprint.plugins.youtubelive - INFO - Start stream command received.
```
When I use the UI to try to start and stop the stream. But the behavior hasnt changed.

When I run the test manually with: 
```
docker run --privileged --name YouTubeLive -ti octoprint/youtubelive:latest http://<my_ip>/webcam/?action=stream <my_stream_id> null
```
The stream starts on youtube just fine. It just doesnt obey the video settings I pick in the webcam settings for flipping the video, so the stream is upside down. The stock OctoPrint Control tab shows the video properly, as the timelapses I've recorded this week have been fine.

Pitta, 2018-06-10:
Seems like the webcam output is a separate issue I'll tackle. Would like to get the UI going for the streaming though. Shelling in to start and stop is less than convenient.

jneilliii, 2018-06-10:
It pulls those settings from the default octoprint options. Screenshot highlighted above.

jneilliii, 2018-06-10:
If your camera looks right on the control tab it will look right on YouTube.

Pitta, 2018-06-10:
Ahh ok great. I have ~60 mins left on this thing. Less if it fails where i've been having trouble. Then I'll be back to try a reboot.

Pitta, 2018-06-10:
Interesting - that seemed to work!
I also edited my octoprint.txt config settings to set the camera correctly before the server boots up. Much better now. 

Thanks for the help.

jneilliii, 2018-06-10:
I think the issue might be because the plugin gets the stream url on startup of octoprint, would have to go back and look through the code to verify.  

The stream url has to have the full url because docker runs the stream re-encoding in it's own virtual network or something like that...sorry not much help there as I'm not super familiar with how it works, but I found posts about it elsewhere where referencing 127.0.0.1 actually resolves back to the docker instance itself and not the pi that it's running on.

jneilliii, 2018-06-10:
Ok, so I just went back and looked at the code and it should pull the stream url right when you start the stream, but I think OctoPrint caches the setting or something.  Just restarting OctoPrint to re-load the settings would probably have resolved the issue in lieu of a full system reboot after making the stream url change.

Pitta, 2018-06-10:
Thanks for all your help.