jneilliii/OctoPrint-YouTubeLive #29: YouTube stream id ?

bonvalletRomain, 2018-06-18:
Hello,
I can't find the YouTube Stream ID. Could you show me where this element is located ?

jneilliii, 2018-06-18:
Did you follow the instructions [here](https://github.com/jneilliii/OctoPrint-YouTubeLive/blob/master/docker_instructions.md).  Otherwise, you're going to have other issues.  

You can find your stream id on your [YouTube Live Dashboard](https://www.youtube.com/live_dashboard) page. just press the reveal button.

[image]

bonvalletRomain, 2018-06-19:
I have this message when i follw the instruction

`pi@octopi:~ $ docker pull alexellis2/streaming:17-5-2017
17-5-2017: Pulling from alexellis2/streaming
a5cf18dd7180: Pulling fs layer 
4b7dd464ea4e: Pulling fs layer 
5bb79798d0a5: Pulling fs layer 
65207fdf017c: Waiting 
979101378ced: Waiting 
ac08c15aa62d: Waiting 
3fc53e81bdbf: Waiting 
3d534a4386c7: Waiting 
9fb590bfb52b: Waiting 
b9095b98b91b: Waiting 
89140f653e19: Waiting 
bf03a411b030: Waiting 
0bc6ac44347c: Waiting 
fc7767506cd5: Waiting 
80c12aad48ff: Waiting 
af84ae0105dd: Waiting 
2e3c6a14eb15: Waiting 
error pulling image configuration: Get https://production.cloudflare.docker.com/registry-v2/docker/registry/v2/blobs/sha256/63/63e724cc90ea26bde80d3ed6548eb040f5f9a83d01f5683a030a3d31d178b743/data?verify=1529411586-2Qv7VsRl0Tre%2FEYmMJhV9QsMCoo%3D: net/http: TLS handshake timeout`

jneilliii, 2018-06-19:
Looks like alexellis2 updated his docker image for Raspbian Stretch.  Try this command and let me know if it's able to pull and I'll update the instructions accordingly.

```
docker pull alexellis2/streaming:07-05-2018
```

bonvalletRomain, 2018-06-19:
I Have this message 
`pi@octopi:~ $ docker pull alexellis2/streaming:07-05-2018
07-05-2018: Pulling from alexellis2/streaming
b906b103c0d7: Extracting [=============>                                     ]  14.16MB/52.36MB
d2cb0e116c35: Download complete 
bdc397860cef: Download complete 
8d6c3df98ed8: Download complete 
6d96283e4796: Download complete 
251e2212d0b5: Download complete 
168e665517bb: Download complete 
d941c93a358d: Download complete 
17a5c730439f: Download complete 
2dd0d3fc92c5: Download complete 
8c86036bd422: Download complete 
1ecc4deb91bb: Downloading [=============================================>     ]  74.02MB/82.03MB
49800e533d71: Downloading [==============>                                    ]  71.19MB/240.7MB
7961560d62c7: Downloading 
e602239716dd: Waiting 
12fcd33ca01e: Waiting 
e253f062764d: Waiting 
c176879df3d1: Waiting 
f831db4762ef: Waiting 
net/http: TLS handshake timeout`

jneilliii, 2018-06-20:
Thanks.  I just recently installed OctoPi 0.15 and haven't yet installed this plugin.  I'll go through the steps and verify.

jneilliii, 2018-06-20:
So it took me several attempts with this docker pull command to complete in the latest octopi release on my pi zero w. I ended up having to reboot a couple of times before the command would actually complete.  Keep trying until the docker image is successfully downloaded (~640MB) and installed.

StevieMac70, 2019-12-11:
I have set up everything, including updating the webcam stream address, but still get the following error

YouTube Live Error
('Connection aborted.', error(2, 'No such file or directory'))

jneilliii, 2019-12-11:
Did you follow the instructions for creating the docker image, etc.?

StevieMac70, 2019-12-11:
I'm not running OctoPrint in a docker, nor have Docker installed.  I'm using the OctoPi version.  Version 0.16.0, running on Raspberry Pi 3 Model B Plus Rev 1.3

StevieMac70, 2019-12-11:
Working on it now....

jneilliii, 2019-12-11:
Yeah, there's additional steps required for the plugin to work. It looks like you found them.

StevieMac70, 2019-12-11:
Installed the docker and followed the rest of the instructions.  But still no feed.  I got an error in OctoPrint with a "cannot find container".

jneilliii, 2019-12-11:
If you ssh back into the pi and run the command below what is listed? 

```
docker container ls --all
```

StevieMac70, 2019-12-15:
Sorry for my delay.  Been a long last couple of days...

CONTAINER ID        IMAGE                          COMMAND                  CREA                                                                                                             TED             STATUS                    PORTS               NAMES
e40306924cbc        octoprint/youtubelive:latest   "./entry.sh http://1…"   4 da                                                                                                             ys ago          Exited (255) 4 days ago                       YouTubeLive1

jneilliii, 2019-12-15:
That looks correct. I'm confused why it wouldn't be able to run unless something upstream that I'm referencing in the docker build has changed. I kind of decided to abandon this project because the webcam streamer plugin is easier to setup and has more options.

StevieMac70, 2019-12-15:
ok, cool.  Thanks for your help.  I will see if I can find that other plugin.

Webranger2, 2019-12-31:
Why is that?

YouTube Live Error
404 Client Error: Not Found ("No such container: cdd368c2ebb7158aca7d1b7113dd2991c980c62098393040ea9a48012d303e59")