jneilliii/OctoPrint-YouTubeLive #30: ('Connection aborted.', error(2, 'No such file or directory'))

reederda, 2018-06-23:
I installed this plugin from Octoprint's plugin manager. I followed the instructions regarding the channel and stream ids, but when I try to "go live" I get the following error:
('Connection aborted.', error(2, 'No such file or directory'))

jneilliii, 2018-06-23:
Did you change your webcam stream url to include the IP address?

reederda, 2018-06-23:
I have done that, yes.

jneilliii, 2018-06-23:
Can you please upload your octoprint.log file or look for any errors in that file related to the plugin? Some have said that they had to restart octoprint after setting the webcam stream url to include the IP for it to take effect.

reederda, 2018-06-24:
Upon further inspection, it appears that the docker container did not build successfully. I'm trying to now, however, and am receiving the error:

standard_init_linux.go:190: exec user process caused "exec format error"
The command '/bin/sh -c chmod +x entry.sh' returned a non-zero code: 1

gretel, 2018-07-12:
i get that exact same error now - as i re-installed this plugin.

jneilliii, 2018-07-12:
@gretel, did you start from scratch with the setup instructions, or just reinstall the plugin part inside OctoPrint? It was working before and now it doesn't? Is the test command successful from the instructions?

gretel, 2018-07-12:
@jneilliii no, yes, yes, yes. 😁

jneilliii, 2018-07-12:
That is odd. Anything else change outside of uninstalling the plugin and reinstalling? Does the webcam url in OctoPrint settings still have the ip address included?

gretel, 2018-07-16:
i had to postpone further investigation but are trying to get back, soon.

closed 2018-08-06 by stale[bot]

otherguy, 2018-08-29:
Any updates? I have the same issue. Fresh OctoPi/OctoPrint install.

jneilliii, 2018-08-30:
@otherguy, did you follow the prerequisite instructions found [here](https://github.com/jneilliii/OctoPrint-YouTubeLive/blob/master/docker_instructions.md)?

otherguy, 2018-08-30:
@jneilliii I did but it was quite late (or early in the morning) 😉  I'm well versed with Docker and I'll look into it on the weekend.

jneilliii, 2018-08-30:
I know near nothing about docker, so you're in better shape than myself. I just hacked this together the best I could. The one thing that is the most common issue is changing octoprint's webcam and timelapse setting to be a full url with IP because relative pathing won't be interpreted correctly by the docker image. Make sure to restart octoprint after making that change as well.

pstimpel, 2019-01-03:
I have the same error. fresh octoprint install, up to date, installed the plugin. I was verifying my youtube settings using obs, working. I made sure using the FQDN for the stream cam. I restarted both octoprint and the RPI. I receive the same error over and over.

I was trying those for the camera

http://myip/webcam/?action=stream

http://myip:8080/?action=stream

Both URLs show the stream of my camera, however, livestreaming failes with the message from the subject.

octoprint.log 

2019-01-03 21:13:59,771 - octoprint.plugins.youtubelive - INFO - OctoPrint-YouTubeLive loaded! Checking stream status.                                                              

2019-01-03 21:13:59,781 - octoprint.plugins.youtubelive - ERROR - ('Connection aborted.', error(2, 'No such file or directory'))

Any idea?

jneilliii, 2019-01-03:
@pstimpel; did you follow the instructions linked in the [requirements for streaming](https://github.com/jneilliii/OctoPrint-YouTubeLive#requirements-for-streaming) section? Seems like you don't have docker image set-up...you can't just install the plugin there are other steps...

hevilp, 2019-01-26:
Do you can change the plugin to use an existing mjpg_streamer and not the docker image?

jneilliii, 2019-01-26:
The docker image re-encodes the existing mjpegstream to the appropriate format and pushes it to YouTube.