jneilliii/OctoPrint-YouTubeLive #31: Input/output error

Scottmcf, 2018-06-28:
Filter: null
ffmpeg version N-90990-g652b857b94 Copyright (c) 2000-2018 the FFmpeg developers
  built with gcc 6.3.0 (Raspbian 6.3.0-18+rpi1+deb9u1) 20170516
  configuration: --arch=armel --target-os=linux --enable-gpl --enable-omx --enable-omx-rpi --enable-nonfree
  libavutil      56. 18.100 / 56. 18.100
  libavcodec     58. 19.101 / 58. 19.101
  libavformat    58. 13.101 / 58. 13.101
  libavdevice    58.  4.100 / 58.  4.100
  libavfilter     7. 21.100 /  7. 21.100
  libswscale      5.  2.100 /  5.  2.100
  libswresample   3.  2.100 /  3.  2.100
  libpostproc    55.  2.100 / 55.  2.100
Input #0, mjpeg, from 'http://172.17.0.1:8080/?action=stream':
  Duration: N/A, bitrate: N/A
    Stream #0:0: Video: mjpeg, yuvj420p(pc, bt470bg/unknown/unknown), 640x480, 5 tbr, 1200k tbn, 5 tbc
Guessed Channel Layout for Input Stream #1.0 : stereo
Input #1, s16le, from '/dev/zero':
  Duration: N/A, bitrate: 1411 kb/s
    Stream #1:0: Audio: pcm_s16le, 44100 Hz, stereo, s16, 1411 kb/s
rtmp://a.rtmp.youtube.com/live2/xxxx-xxxx-xxxx-xxxx: Input/output error

jneilliii, 2018-06-28:
It appears to me that on the test command that you ran the stream id/key was not entered.  You need to replace the `xxxx-xxxx-xxxx-xxxx` with your personal YouTube stream key.

```
docker run --privileged --name YouTubeLive -ti octoprint/youtubelive:latest http://<ip>:8080/?action=stream xxxx-xxxx-xxxx-xxxx null
```

Scottmcf, 2018-06-28:
It was in there I just x'ed it out to keep it private like it tells you to do.

jneilliii, 2018-06-29:
Gotcha, just needed to make sure.  What day did you follow the steps, because I recently updated the docker pull command reference for Raspbian Jessie and not sure if that might be what's causing the issue here or not. It could also be something related to how the source stream is pulling.  Can you try the following url for the test command and let me know. Also, just to confirm; the ip you're using for input (172.17.0.1) has a high likelihood of being your router's gateway and not your pi's ip address.

`http://172.17.0.1/webcam/?action=stream`

Scottmcf, 2018-06-30:
I installed it on the 27th I believe. may have started on the 26th and stopped on the 27th. I may have replace the ip with a fake one also. I will work on it some more tonight.

jneilliii, 2018-06-30:
That would definitely have been after the updated instructions, but it seems like the ffmpeg command is running, so I would venture to say it's related to one side of the stream since it's an input/output error.  Sorry, I'm not an expert on ffmpeg.

Scottmcf, 2018-06-30:
I got it to work with the test in putty but not with Octoprint.

Scottmcf, 2018-06-30:
docker run --privileged --name YouTubeLive -ti octoprint/youtubelive:latest http://192.168.2.28:8080/?action=stream xxxx-xxxx-xxxx-xxxx null   I did not keep the line from the first night but I noticed I had a typo the second night. In Octoprint when I hit go Live it looks like it starts but youtube does not show that it is streaming. when I click stop I get a 404 error saying no such container found.

jneilliii, 2018-06-30:
Assuming the test command works update your octoprint webcam settings with that same url, restart octoprint and the go live button should work.

Scottmcf, 2018-06-30:
The whole url from docker to null?

Scottmcf, 2018-06-30:
I have been using the first one. Will try the second one and see what happens. this is error with first link.
YouTube Live Error
404 Client Error: Not Found ("No such container: 48b9b29ff15cb8a9e3f360201b39a1cfa2df9bd15bf8

Ok. I reset my stream key and put the new one in Octoprint and it finally works. I tried both the url's you posted and they both work. not sure what I did to mess up the key but something was wrong with it. 
Thanks for all the help.

closed 2018-06-30 by Scottmcf

Scottmcf, 2018-07-02:
How long should the stream run? After about 24 minutes it shuts down.

jneilliii, 2018-07-02:
There are no limits imposed by the plugin. Not really sure why it would stop unless it's a processing issue. What model pi are you running on?

Scottmcf, 2018-07-02:
It is the 3 B. I will try a different router and see if that changes anything. I know my laptop reports it lost internet but I always blamed the laptop.

jneilliii, 2018-07-03:
Nice, glad it worked out for you.