jneilliii/OctoPrint-YouTubeLive #32: Could This Work For Twitch

BeerGeekGamer, 2018-08-09:
Hello,

I was wondering if you have any insight if this could work with Twitch. I read the blog article which the work is based on, and even though I do have some programming skills, I will admit this is a little more advance then the skills I possess. I am not familiar with Docker, but could you edit alexellis2 image to point to the Twitch rtmp location instead of YouTube's?

jneilliii, 2018-08-09:
Assuming twitch uses a similar rtmp url to post to I would think it would work just changing the url in the entry.sh file would do the trick. I'll see about creating a twitch account and test for you.

BeerGeekGamer, 2018-08-09:
That would be awesome!

I have figured out a little work around, by installing libav-tools and using the following command:

`raspivid -t 0 -fps 25 -hf -b 2000000 -o - | avconv -i - -vcodec copy -an -r 30 -g 30 -bufsize 2000000 -pix_fmt yuv420p -f flv rtmp://TWITCH_INGEST_ENDPOINT/YOUR_STREAM_KEY_HERE`

The only thing is, you have to kill the mjpg_streamer process for this to work, or you'll get `mmal_vc_component_enable failed to enable component enospc` since mjpg-streamer is started at boot and is using the camera. I was just hoping you could do this with mjpg-streamer since it's already installed through Octopi and then I wouldn't have to install anything else.

jneilliii, 2018-08-09:
I'll probably end up forking this plugin to a new one in order to have the right view tab, as that would definitely not work. Do you happen to know if twitch is like youtube and has a direct url to view a specific user's stream?

BeerGeekGamer, 2018-08-09:
Yes, for Twtich you can use this site ([https://stream.twitch.tv/ingests/](url)) to find the recommended Twitch rtmp endpoints, based on the optimal network paths from your device. Then from your Twitch channel you grab our Stream Key and just append it to the end.

So for example: rtmp://live-dfw.twitch.tv/app/YOUR_TWITCH_KEY_HERE

jneilliii, 2018-08-10:
[image]

closed 2018-08-10 by jneilliii

BeerGeekGamer, 2018-08-10:
> I think I have a working generic RTMP streamer that is working for me with
> my tests on Twitch. The process is similar to this plugin to get up and
> running. Follow the instructions over here
> To install the plugin side use the following url in plugin manager once
> you've followed those instructions.
>
> https://github.com/jneilliii/OctoPrint-RTMPStreamer/archive/master.zip
>
> You may not need to install docker if you already have it installed. The
> docker build took me several attempts to complete as my pi kept timing out
> for some reason, but keep trying the docker build line until it completes.
>
> Once everything is successfully tested enter your full stream url, view
> url and resolution in the RTMP Streamer Settings. For resolution I don't
> think anything greater than 640x480 would be beneficial unless your mjpg
> stream is at a higher resolution than mine. ffmpeg seems to detect the
> stream at that level for me and I use a default octopi install. For twitch
> you can use the following info...
>
> stream url: rtmp://live-dfw.twitch.tv/app/YOUR_TWITCH_KEY_HERE
> view url: https://player.twitch.tv/?channel=beergeekgamer&muted=true or
> http://player.twitch.tv/?channel=beergeekgamer&muted=true based on which
> protocol you access your octoprint on (http vs https).
>
> If you run into issues please post over on the new plugin's issue tracker
> here <https://github.com/jneilliii/OctoPrint-RTMPStreamer/issues>.

Awesome, thank you so much. I don't know if I'll have time to test it out
today, but I'll definitely let you know once I get it up and running.