jneilliii/OctoPrint-YouTubeLive #33: Stream quality very low (choppy, lots of compression / artifacts)

rlueckl, 2018-08-12:
The title says it pretty much. I've just installed the plugin (thanks for making it!) and setting it up was pretty easy, but the stream quality is really low. It's choppy and I see a lot of compression and it's lost a lot of detail (blurry edges, etc.). I have a 40Mbps upload and often stream 720p without any problems, so I guess it's not my connection, rather the settings of this plugin.

Can you recommend better settings (no compression at all, if possible)? I have OctoPrint installed on a RasPi 3 Model B and my camera is a PlayStation Eye (PS3 camera).

See the picture for comparison between the YouTube stream and the direct feed from OctoPrint.

[image]

jneilliii, 2018-08-12:
I may have a solution for this. I'm testing my generic rtmpstreamer plugin and added resolution and frame rate options for it. Will eventually get around to adding those back into this plugin.

rlueckl, 2018-08-13:
That sounds great! Thanks!

rlueckl, 2018-08-15:
Thanks for the tip! I don't have time now, but I'll try it on the weekend.

closed 2018-09-05 by stale[bot]