jneilliii/OctoPrint-YouTubeLive #34: "At least one output file must be specified."

xXRoboBoyXx, 2018-08-20:
I run the command 
"sudo docker run --privileged --name YouTubeLive -ti octoprint/youtubelive:latest http://192.168.1.124:8080/?action=stream dont-want-to-show" and i get a "At least one output file must be specified" error.

"OctoPrint Webcam Stream: http://192.168.1.124:8080/?action=stream
Live-stream secret: dont-want-this-shown
Filter:
ffmpeg version N-90990-g652b857b94 Copyright (c) 2000-2018 the FFmpeg developers
  built with gcc 6.3.0 (Raspbian 6.3.0-18+rpi1+deb9u1) 20170516
  configuration: --arch=armel --target-os=linux --enable-gpl --enable-omx --enable-omx-rpi --enable-nonfree
  libavutil      56. 18.100 / 56. 18.100
  libavcodec     58. 19.101 / 58. 19.101
  libavformat    58. 13.101 / 58. 13.101
  libavdevice    58.  4.100 / 58.  4.100
  libavfilter     7. 21.100 /  7. 21.100
  libswscale      5.  2.100 /  5.  2.100
  libswresample   3.  2.100 /  3.  2.100
  libpostproc    55.  2.100 / 55.  2.100
Trailing options were found on the commandline.
Input #0, mjpeg, from 'http://192.168.1.124:8080/?action=stream':
  Duration: N/A, bitrate: N/A
    Stream #0:0: Video: mjpeg, yuvj420p(pc, bt470bg/unknown/unknown), 640x480 [SAR 1:1 DAR 4:3], 5 tbr, 1200k tbn, 5 tbc
Guessed Channel Layout for Input Stream #1.0 : stereo
Input #1, s16le, from '/dev/zero':
  Duration: N/A, bitrate: 1411 kb/s
    Stream #1:0: Audio: pcm_s16le, 44100 Hz, stereo, s16, 1411 kb/s
At least one output file must be specified

It stops there, and won't do anything else. Any help would be appreciated.

jneilliii, 2018-08-20:
I think your test command is just missing the null at the end?

xXRoboBoyXx, 2018-08-20:
*facepalm* Thank you, It's fine now.

xXRoboBoyXx, 2018-08-20:
Or so i thought, Now i get a permission denied while pressing "Go Live"

jneilliii, 2018-08-20:
That makes me think you forgot to do the line in the instructions to give the pi user permissions to docker.

xXRoboBoyXx, 2018-08-20:
That was the second command i typed, i for sure put it in.

jneilliii, 2018-08-20:
Ok, did you remove the image after running the test?

jneilliii, 2018-08-20:
Sorry, the last part of the test section...the ctrl+c is to stop the running process the second command is to remove the docker image from from the queue so the plugin can add it back in when you press Go Live.

```
ctrl+c
docker rm YouTubeLive
```

xXRoboBoyXx, 2018-08-20:
I've done that command before, "Go Live" doesn't change after i type that commend.

jneilliii, 2018-08-20:
Ok, a couple of other things to verify...test the following command and verify it streams.

```
docker run --privileged --name YouTubeLive -ti octoprint/youtubelive:latest http://192.168.1.124/webcam/?action=stream xxxx-xxxx-xxxx-xxxx null
```

if all is good there then run this.

```
ctrl+c
docker rm YouTubeLive
```

Then make sure to enter the same source url in your OctoPrint webcam & timelapse settings so that it has the full url like below and restart OctoPrint.

```
http://192.168.1.124/webcam/?action=stream
```

xXRoboBoyXx, 2018-08-21:
It's working now, thanks.

closed 2018-08-21 by jneilliii