jneilliii/OctoPrint-YouTubeLive #36: Docker cannot connect to localhost port

sadegroo, 2018-09-12:
In the documentation the docker instance is called like this: 

`docker run --privileged --name YouTubeLive -ti octoprint/youtubelive:latest http://127.0.0.1:8080/?action=stream XXXXX`

However, docker denies access to the localhost of the host itself by default. This can be bypassed with the `--network="host"` flag. More information can be found [here](https://stackoverflow.com/a/24326540).

The full docker command would be:

`docker run --privileged --network="host" --name YouTubeLive -ti octoprint/youtubelive:latest http://127.0.0.1:8080/?action=stream XXXXX`

If possible edit this line in the code:
https://github.com/jneilliii/OctoPrint-YouTubeLive/blob/707376798a2605e50c93ce7fd6710e7aac346166/octoprint_youtubelive/__init__.py#L97

If you add there the network and host flag, it would work for localhost. Thanks!

jneilliii, 2018-09-12:
Thanks so much for that.  It was driving me nuts why I couldn't get docker to connect to localhost (don't know much about docker).  I never updated the instructions, but told people to use the actual ip of the machine rather than the localhost loopback.  They would have to also update that in the webcam & timelapse settings to include the full url with real ip instead of localhost.

jneilliii, 2018-09-12:
Merged change from @sadegroo's fork into master, will update version number etc. later and push an official version release.

sadegroo, 2018-09-12:
It appears that it should be `network_mode` instead of `network` like mentioned here: https://docker-py.readthedocs.io/en/stable/containers.html
[image]

sadegroo, 2018-09-13:
My friend Vincent (@vincentcox) actually found this issue while he was setting up nginx on my RPI's and your pluging "broke".
I still need to prepend the local IP in octoprint's webcam stream setting though.

vincentcox, 2018-09-13:
I created a [fork](https://github.com/vincentcox/OctoPrint-YouTubeLive). @sadegroo, will test this fork on his OctoPrint instance and if it works I will create a pull request. This makes it more clear for all of us.

jneilliii, 2018-09-13:
Yeah, thanks @vincentcox and @sadegroo.  I see both your forks and have incorporated the changes into the networ_mode_test branch.  About to test it myself to verify functionality.

jneilliii, 2018-09-13:
One thing that could be done is to hard code the input stream to the localhost address, but it seems like that wouldn't be ideal as not everyone uses the same set-up etc. For example, you using NGINX vs me using OctoPi's built in haproxy vs others using Windows and Yawcam, etc.  Keeping the option to read from the stream url setting still makes the most sense, just have to make sure the full url is used so the ffmpeg command can "see" it.  By the way, how was this broken with NGINX, just curious what this fixed overall that was the issue, for documentation purposes.

sadegroo, 2018-09-14:
Couldn't you search the stream URL for a leading "/", or "http" and then assume local host or full URL accordingly ?

Btw my day job is humble PLC programming, so i'm no python expert (yet).

@vincentcox will have to inform you about the nginx issues.

jneilliii, 2018-09-14:
Yeah @sadegroo, that would be a possibility.  BTW...I'm just an IT manager with an Architecture (as in building design) background. All the python programming I have learned has been to program plugins for OctoPrint.  Great learning experience.

closed 2019-08-09 by jneilliii