jneilliii/OctoPrint-YouTubeLive #40: Unable to Reslove Host

aw-studios, 2018-10-02:
Hi, I have wanted to install this plugin and when I get to the second step, sudo usermod pi-aG docker. It gives me "unable to resolve host ______" I don't know what is going on.
[image]

aw-studios, 2018-10-03:
Let me take a look at that.

aw-studios, 2018-10-03:
That's what I put into the pi. It was just a typo. It works though. I just want to know if this is needed for running the program.

jneilliii, 2018-10-03:
All the steps listed on the [streaming requirements instructions](https://github.com/jneilliii/OctoPrint-YouTubeLive/blob/master/docker_instructions.md) are necessary for the plugin to work correctly.

aw-studios, 2018-10-04:
Well, when I put it in, It doesn’t work for me.

jneilliii, 2018-10-04:
Ok, I think I see what's going on now.  You are connecting to SSH via hostname vs ip.  If you connect via ip instead and run that command does it give you the same error?

aw-studios, 2018-10-04:
I am actually running it through VNC. I have tried to SSH and I have gotten the password wrong for some reason, even though I have reset it a couple time. Also, what do you mean by IP? I am pretty new to the whole Raspberry Pi and coding universe.

jneilliii, 2018-10-04:
I'll have to test that with VNC, never done it that way before, I'll try tonight when I get home.  IP is an abbreviation for Internet Protocol, and I should have said `IP address`, which is the address assigned to any device when on a network. For example your modem/wireless router might have an IP address of 192.168.0.1 and your Pi have an IP address of 192.168.0.2.

aw-studios, 2018-10-05:
Okay. I got what IP means, but, what do you mean IP vs SSH? My SSH doesn’t work so how would you access with IP?

closed 2018-10-26 by stale[bot]