jneilliii/OctoPrint-YouTubeLive #41: Not working for me, must be my fault

jaxxtheone, 2018-10-10:
Error message when starting to print:, on htt://octopi.local

**YouTube Live Error
409 Client Error: Conflict ("Conflict. The container name "/YouTubeLive" is already in use by container "f4742833f895c073742344c2507635369292969addb9b49e875d1adba0d82e82". You have to remove (or rename) that container to be able to reuse that name.")**

Tried several approaches but didn´t get success.

jneilliii, 2018-10-10:
That indicates you didn't remove the container during testing maybe. SSH to your pi and type docker rm YouTubeLive.

jaxxtheone, 2018-10-10:
DId that before your answer. when I do that I can´t go live. I didn´t understand where to put the URL:
**http://192.168.2.98:8080/?action=stream** so I can get it on air.
Before removing had a bunch of answers like that: 
**frame= 1219 fps=5.0 q=-0.0 size=   21072kB time=00:04:03.40 bitrate= 709.2kbits/s [mjpeg @ 0x2346540] unable to decode APP fields: Invalid data found when processing input
    Last message repeated 2 times
frame= 1222 fps=5.0 q=-0.0 size=   21109kB time=00:04:04.00 bitrate= 708.7kbits/s [mjpeg @ 0x2346540] unable to decode APP fields: Invalid data found when processing input
    Last message repeated 1 times**
Sorry, too nooby on this side.

jaxxtheone, 2018-10-10:
And for that matters:
```
pi@octopi:~/youtubelive $ ls -a
.  ..  Dockerfile  entry.sh  .git
```

jneilliii, 2018-10-10:
Try going into octoprint settings, webcam & timelapse and put that url in the stream url field.

jaxxtheone, 2018-10-11:
Tried that. Still no joy.
Tried to redo the whole process as bellow:
`

pi@octopi:~ $ docker run --privileged --name YouTubeLive -ti octoprint/youtubelive:lates http://192.168.2.98:8080/?action=stream h1a0-5dw6-j0ha-90pz
Unable to find image 'octoprint/youtubelive:lates' locally
docker: Error response from daemon: pull access denied for octoprint/youtubelive, repository does not exist or may require 'docker login'.
See 'docker run --help'.
pi@octopi:~ $ cls
-bash: cls: command not found
pi@octopi:~ $ clear
pi@octopi:~ $ docker pull alexellis2/streaming:07-05-2018
07-05-2018: Pulling from alexellis2/streaming
Digest: sha256:ef275dd996fd941a70efeda19267bf0da86e98f35843f62f27f1efe4d8b849fa
Status: Image is up to date for alexellis2/streaming:07-05-2018
pi@octopi:~ $ cd ~
pi@octopi:~ $ git clone https://github.com/jneilliii/youtubelive --depth 1
fatal: destination path 'youtubelive' already exists and is not an empty directory.
pi@octopi:~ $ cd youtubelive
pi@octopi:~/youtubelive $ docker build -t octoprint/youtubelive .
Sending build context to Docker daemon  54.78kB
Step 1/3 : FROM alexellis2/streaming:07-05-2018
 ---> a8d02138647f
Step 2/3 : COPY entry.sh entry.sh
 ---> Using cache
 ---> f344d8b4c6db
Step 3/3 : RUN chmod +x entry.sh
 ---> Using cache
 ---> 59b282798b85
Successfully built 59b282798b85
Successfully tagged octoprint/youtubelive:latest
pi@octopi:~/youtubelive $ docker run --privileged --name YouTubeLive -ti octoprint/youtubelive:lates http://192.168.2.98:8080/?action=stream xxxx-xxxx-xxxx-xxxx
Unable to find image 'octoprint/youtubelive:lates' locally
docker: Error response from daemon: pull access denied for octoprint/youtubelive, repository does not exist or may require 'docker login'.
See 'docker run --help'.
pi@octopi:~/youtubelive $
`

jaxxtheone, 2018-10-11:
Tried all "Test" buttons in Webcam and Timelapse and the three were successfully. 
By the way changed the xxxx-xxxx-xxx-xxxx string as it was aired and not intended to. 
So, on my side there´s something fishy. Maybe, as tomorrow is day-off, I´ll try to create a new octopi image and start from skratch.
I will post again my progress. asap

jaxxtheone, 2018-10-16:
Just restarted my effort, hold on. Will finish my print  and try tomorrow fix my plugin.

jaxxtheone, 2018-10-17:
It was faulty SD. That´s working, sorry for the inconvenience,

closed 2018-10-17 by jaxxtheone

jneilliii, 2018-10-17:
No problem, glad you got it worked out.