jneilliii/OctoPrint-YouTubeLive #44: Not streaming

adilinden, 2019-01-17:
Hi,

I have streaming working successfully. However, every so often the stream won't start. When I stop the stream I get the following message:

> YouTube Live Error
404 Client Error: Not Found ("No such container: .....

Below is a screenshot.

[image]

Now, this error only pops up when stopping the stream. Trying to start the stream fails, then stopping the stream pops up another error. When I reboot the Raspberry Pi all works well once it comes back up. Do note, that all other functions are working just fine regardless of the plugin error.

Here is what docker has to say:

```
pi@octopi:~ $ docker image ls
REPOSITORY              TAG                 IMAGE ID            CREATED             SIZE
octoprint/youtubelive   latest              95639181f1a0        6 days ago          1.32GB
alexellis2/streaming    07-05-2018          a8d02138647f        8 months ago        1.32GB
```
```
pi@octopi:~ $ docker container ls
CONTAINER ID        IMAGE               COMMAND             CREATED             STATUS              PORTS               NAMES
```
I cannot find anything in the log files in `/var/log/`

Thanks,
Adi

jneilliii, 2019-01-18:
Yeah, stopping the stream would result in an error if the docker container never started when you run from the plugin.  I would be curious as to why that isn't happening.  Does it seem to do it more after the pi has been running for a while? Any errors in octoprint.log?

I honestly don't use the plugin because of the stream delay and poor video quality that I could never resolve on a low powered pi on wifi.

adilinden, 2019-01-19:
It happened when I was initially setting it up and then I just assumed it was something I missed or did wrong. Eventually a reboot resolved that. After that I spent several days printing and streaming, including a 31 hour stream. A subsequent short streamed events worked just fine, until finally the stream that failed. There was nothing I could do to convince it to return to working properly. Unfortunately I do not think I tried a manual stream from CLI. Although looking at the docker images and containers all looked normal (as mentioned in my original note). 

I did not make any changes to the docker images at all. A reboot resolved everything.

Unfortunately, I am no longer using the plugin in its original form. I forked it and made a "few" changes. You're welcome to see if what I did is of value and worth merging. Note that only once I merged into another project, and that what with very clear instructions ([pid.codes](http://pid.codes/)). I will need some help if you're interested in pulling in the changes.

My fork of your project is at [adilinden-oss/octoprint-youtubelive](https://github.com/adilinden-oss/octoprint-youtubelive).

What I accomplished is:

- Configure webcam URL in plugin settings independent of global setting (I am accessing OctoPi direct and via reverse proxy. For reverse proxy to work the global needs to be relative.)
- Small docker container of 60MB
    ```
    pi@octopi:~ $ docker image ls -a
    REPOSITORY             TAG                 IMAGE ID            CREATED             SIZE
    adilinden/rpi-stream   latest              ccc141d4e884        6 hours ago         360MB
    ```
- Able to get `bash` in container (troubleshooting)
- Container pulled not built

Sorry for this latter part being off-topic of the original issue.

adilinden, 2019-01-20:
Having bash access to the container directly allows for running ffmpeg directly without having to make any changes to the Dockerfile. I just find it easier that way. But more so, being able to do a `docker pull` was of more importance to me.

As for the smaller container, I used a multistage build. The build creates two images, the first image installs all the build tools and does the compiling, the second image is the runtime image that during build copies only the results from the build image into itself. The resulting image won't have any of the bloated build tools and other artifacts presents.

For what it's worth I am still pretty fresh with docker as well. Spent a lot of time understanding it, but only slowly seeing the full potential of it.

- [Dockerfile](https://github.com/adilinden-oss/docker-rpi-ffmpeg/blob/master/Dockerfile)
- [Docker Docs - Use multi-stage builds](https://docs.docker.com/develop/develop-images/multistage-build/)

adilinden, 2019-01-22:
The original issue is still happening. I don't think it has anything to do with `docker`. I can manually start and stop the stream by creating and tearing down a running `docker` container.

While printing is going on, even when I turn the stream off it is automatically restarted (I have that checkbox checked in settings). I waited till print was finished, then stopped the stream and tried starting it. The stream never starts. No log entries beyond the `Start stream command...`.

```
2019-01-22 17:20:33,377 - octoprint.plugins.youtubelive - INFO - Stop stream command received.
2019-01-22 17:20:57,744 - octoprint.plugins.youtubelive - INFO - Start stream command received.
```

This time I only restarted the `octoprint` server, not the entire Raspberry Pi. Now streaming works again. For now I don't know enough about the octoprint internals, plus it takes days for it to break...

Tjololo12, 2019-01-28:
I'm experiencing the same issue. I'm able to run the docker container manually, but I am unable to automatically start the stream on the octoprint tab. It just says "Start stream command received". I'm not sure where to look for logs.

nionio6915, 2019-02-01:
I found this thread after figuring out that a reboot was needed to restore the stream. 

jneilliii- its actually really helpful to have this feature. i was able to get help in my issue by letting a mentor witness my print as it happened. 
yes, the low resolution and low framerate could be better, but i am grateful for what it is!

adilinden- can your fork be patched over the original install by the same instructions? i'd be curious to try your improvements. but i dont have the linux or python foo many of you do.

jneilliii, 2019-02-01:
@Tjololo12, did you set your webcam & timelapse > stream url setting to include the full ip address and restart octoprint?

[image]

@adilinden, I saw you switched over to https://github.com/adilinden-oss/octoprint-webcamstreamer.  That is very much like my RTMPStreamer plugin, but I do like your changes for the ffmpeg command tokens, very nice.

Tjololo12, 2019-02-01:
@jneilliii I've tried a few different things. My octoprint is on a domain with an HTTPS certificate (served through haproxy). When I try the local IP, my webcam does not work at all: https://i.imgur.com/4cI6WQP.png When I set it to my domain with https, it works just fine: https://i.imgur.com/oAuFyN7.png However, YouTubeLive will not start streaming. When I click "Go Live" it will eventually look like it started: https://i.imgur.com/YLlVjYk.png but on my youtube page it's not actually running: https://i.imgur.com/AgN0QTF.png. Then when I click stop, I get the 404 error showing that the Docker container never actually started: https://i.imgur.com/gvki2En.png

I've narrowed down a few things while testing. First, I know that when I run the docker container manually via the commandline with the domain name and HTTPS, it says that ffmpeg isn't compiled with ssl enabled (or something like that, I can't log into the commandline now to test, I'll try to do so this evening if you need me to), and it doesn't work. However, if I use the IP address (http://192.168.1.179), it appeared to work. However, that URL doesn't work in Octoprint, I assume because I'm visiting via the domain and it's trying to connect from my remote computer to <webcam URL>, which obviously doesn't work because I'm not on the local network.

I did try adilinden's fork, but it didn't appear to work either so I didn't fiddle with it much.

I assume my issue is likely a local configuration issue, but hopefully I can get it figured out. I had a few people who liked watching my prints go, and this would be a cool way to do it.

adilinden, 2019-02-01:
Hi @jneilliii 

Yes, I put a few ideas together in [adilinden-oss/octoprint-webcamstreamer](https://github.com/adilinden-oss/octoprint-webcamstreamer). Having all of that flexibility makes streaming to locations other than YouTube "trivial", although at the same time it allows for things to be badly misconfigured too.

I did rewrite quite a bit of the main plugin and I am cautiously optimistic that it is no longer hanging and requiring a reboot.

So far I had one dropped stream but was able to recover by simply restarting the stream. I am thinking of placing an option into the docker container that restarts `ffmpeg` when it drops until the container is explicitly stopped. I am also hoping to get back to adding a reset button to some of the settings. However, I ran into trouble with that. I saw your comment on my request for help. Unfortunately I was rather disorganized when I attempted to write it and it was lumped in with other changes. So I undid the rest button attempt to salvage that evenings work. I will have to try again and put things into a proper development branch to review.

Adi

jneilliii, 2019-02-01:
@Tjololo12, yeah this is a tricky situation that I believe had a discussion about over on the community forum.  I'll have to see if I can dig that up.  However, if you use @adilinden's you can actually set the source url for the stream separate and have it work.  You would just need to use the local ip address without the webcam folder, but use what you have configured in haproxy for redirecting the /webcam/ folder to.  For example in @adilinden's plugin settings you would set the url to http://192.168.1.179:8080/?action=stream.

Tjololo12, 2019-02-01:
I'll have to give that plugin a try then, thanks! If you do find the discussion let me know, I can move this to there instead of clogging up your github with my own special cases.

jneilliii, 2019-02-01:
This was the discussion I was thinking about...not exactly related as it's specific to redirects, but may be the case in your situation.

https://discourse.octoprint.org/t/ssl-port-forwarding-with-acl-video/4012/16

Tjololo12, 2019-02-02:
I think I got it figured out with @adilinden's plugin. Now I just need to figure out how to increase the bitrate. Thanks for all the help!

closed 2019-02-23 by stale[bot]