jneilliii/OctoPrint-YouTubeLive #47: Docker container fails

hevilp, 2019-01-26:
Hi get this when I follow your instructions:

```
dietpi@DietPi:~/youtubelive$ docker run --privileged --name YouTubeLive -ti octoprint/youtubelive:latest http://192.168.179.220:8080/?action=stream mc4m-ma9w-vtk2-ahs8 null
OctoPrint Webcam Stream: http://192.168.179.220:8080/?action=stream
Live-stream secret: xxxxxxxxxxxxxxx
Filter: null
ffmpeg version N-90990-g652b857b94 Copyright (c) 2000-2018 the FFmpeg developers
  built with gcc 6.3.0 (Raspbian 6.3.0-18+rpi1+deb9u1) 20170516
  configuration: --arch=armel --target-os=linux --enable-gpl --enable-omx --enable-omx-rpi --enable-nonfree
  libavutil      56. 18.100 / 56. 18.100
  libavcodec     58. 19.101 / 58. 19.101
  libavformat    58. 13.101 / 58. 13.101
  libavdevice    58.  4.100 / 58.  4.100
  libavfilter     7. 21.100 /  7. 21.100
  libswscale      5.  2.100 /  5.  2.100
  libswresample   3.  2.100 /  3.  2.100
  libpostproc    55.  2.100 / 55.  2.100
[mjpeg @ 0x2807f60] unable to decode APP fields: Invalid data found when processing input
Input #0, mjpeg, from 'http://192.168.179.220:8080/?action=stream':
  Duration: N/A, bitrate: N/A
    Stream #0:0: Video: mjpeg, yuvj422p(pc, bt470bg/unknown/unknown), 640x480, 5 tbr, 1200k tbn, 5 tbc
Guessed Channel Layout for Input Stream #1.0 : stereo
Input #1, s16le, from '/dev/zero':
  Duration: N/A, bitrate: 1411 kb/s
    Stream #1:0: Audio: pcm_s16le, 44100 Hz, stereo, s16, 1411 kb/s
Stream mapping:
  Stream #0:0 -> #0:0 (mjpeg (native) -> h264 (h264_omx))
  Stream #1:0 -> #0:1 (pcm_s16le (native) -> aac (native))
Press [q] to stop, [?] for help
[mjpeg @ 0x2808540] unable to decode APP fields: Invalid data found when processing input
[swscaler @ 0x288cbd0] deprecated pixel format used, make sure you did set range correctly
[h264_omx @ 0x286b6a0] Using OMX.broadcom.video_encode
[h264_omx @ 0x286b6a0] OMX_GetHandle(OMX.broadcom.video_encode) failed: 80001005
Error initializing output stream 0:0 -- Error while opening encoder for output stream #0:0 - maybe incorrect parameters such as bit_rate, rate, width or height
Conversion failed!
```

jneilliii, 2019-01-26:
Are you running this on an octopi install?

hevilp, 2019-01-26:
no, dietpi with my own octoprint install, why?

jneilliii, 2019-01-26:
Best I can tell from the error below, ffmpeg is complaining about an issue with the hardware acceleration encoder. Doing a quick search of the error I see this specifically for dietpi.

https://dietpi.com/phpbb/viewtopic.php?t=2750
```
[h264_omx @ 0x286b6a0] OMX_GetHandle(OMX.broadcom.video_encode) failed: 80001005
```

hevilp, 2019-01-26:
I think this is the problem [mjpeg @ 0x2807f60] unable to decode APP fields: Invalid data found when processing input

jneilliii, 2019-01-26:
If that's the faulting error then there is a difference with your input stream, which is the mjpgstreamer install/configuration. versus how octopi has it set up.

closed 2019-02-16 by stale[bot]