jneilliii/OctoPrint-YouTubeLive #52: buffering on youtube live every few seconds, but local stream is perfect

yamlCase, 2019-07-21:
OctoPrint 1.3.11 running on OctoPi 0.16.0 
Logitech C920

Stream on youtube plays for about 5 seconds, and buffers for 3 seconds.  the local stream looks fantastic, no buffering at all.  What are some things I can check?

jneilliii, 2019-07-21:
Have you tried using an ethernet connection instead of wifi?  The problem mainly comes form the plugin doing a re-encoding of the mjpgstreamer provided stream of octopi using ffmpeg.  It appears you are using a 4 core device, so processing power shouldn't be an issue like some other users that were having this issue from a zero or older device.  Best I can tell you is to try and get a non-buffering  ffmpeg command line for the mjpg stream working and then update the docker image with that command line.

yamlCase, 2019-07-21:
wifi and wired ethernet behave the same with the buffering.  Thanks for reminding me we build your image on the pi, I'll fiddle around with the entry.sh command line and let you know if something works.

jneilliii, 2019-07-21:
You might want to look into the webcam streamer plugin. I think it allows you to modify the ffmpeg setting directly without rebuilding.

closed 2019-08-12 by stale[bot]