jneilliii/OctoPrint-YouTubeLive #55: why it need entire docker image?

bkrajendra, 2020-02-18:
As I know OctoPi comes with ffmpeg already. Then why it need entire docker image?
Cant we just use ffmpeg directly to stream video.

something like this:

```
ffmpeg -thread_queue_size 512 -f v4l2 -i /dev/video0 \
  -ar 44100 -ac 2 -acodec pcm_s16le -f s16le -ac 2 -i /dev/zero -acodec aac -ab 128k -strict experimental \
  -aspect 16:9 -vcodec h264 -preset veryfast -crf 25 -pix_fmt yuv420p -g 60 -vb 820k -maxrate 820k -bufsize 820k -profile:v baseline \
  -r 30 -f flv "rtmp://a.rtmp.youtube.com/live2/youtube-secret-key
```

jneilliii, 2020-02-18:
You could try. I'm not sure the ffmpeg included with the octopi image includes the advantage of gpu processing, so would potentially be more cpu intensive.

bkrajendra, 2020-02-18:
I tried above command, but getting error "/dev/video0" busy. I dont know how to use camera which being used by Octopi already. I tried disabling camera from Interface but still it shows resource busy.

jneilliii, 2020-02-18:
Yep, you'd have to use the mjpg steam ad input not video directly, unless you stop the webcam service bundled with octoprint, then you don't get the camera in octoprint.

bkrajendra, 2020-02-18:
Hurray !!!!!!!!
Success!!!!!!!!!!!!!!!!!!!!!!!!
[image]

bkrajendra, 2020-02-18:
used command:
```
ffmpeg -re -f mjpeg -i "http://iocareprinter/webcam/?action=stream" -f lavfi -i anullsrc -c:v libx264 -g 60 -c:a aac -ar 44100 -ac 2 -f flv "rtmp://a.rtmp.youtube.com/live2/130e-y458-h08z-ccdr"
```

currently youtube complaining about not getting enough stream but some more tweek in command parameter can help.

ProfEngr, 2020-02-27:
@bkrajendra You pasted your YT credentials. Might want to remove them.

bkrajendra, 2020-02-28:
Thank you for informing @ProfEngr 
I have already deleted test stream.

closed 2020-03-08 by jneilliii

jneilliii, 2021-02-24:
I've made a little progress on decoupling the plugin from using docker. So as long as your device has ffmpeg (recent octopi versions have this be default), then you can try this new version. It gives full control of the ffmpeg commands with default options coming over from the existing docker entry.sh file. I haven't yet dealt with the filters (flip rotate options) but those can be added via command line too. This will allow you to adjust the framerate yourself. 

Install by copying/pasting the URL below into Plugin Manager > Get More > ...from URL.

```
https://github.com/jneilliii/OctoPrint-YouTubeLive/archive/develop/pure_ffmpeg.zip
```

I haven't tested this plugin fully, but very quick tests confirm that the plugin is able to stream to YouTube. The important part is that you have the ffmpeg command configured in OctoPrint's Timelapse Recording settings section.

[image]