jneilliii/OctoPrint-YouTubeLive #58: Only manual streaming with new youtube studio format

T3rmin8torX, 2020-06-09:
Since youtube has switched over to a new creator studio, the plugin is no longer able to auto create video feeds. You have to go to youtube, start a live feed there, then the video feed will be pulled in. In this new format, the stream ID changes with each new manual creation.

jneilliii, 2020-06-09:
What a pain. I no longer really want to support this plugin because of these external dependencies changing and it never really worked as well as direct streaming anyway. Sorry, just don't have the time anymore. May reach out on the community forum/discord server to see if anyone wants to take it over.

jneilliii, 2020-06-09:
Do you have any links for more information related to this change?

T3rmin8torX, 2020-06-09:
Thanks for your hard work up to this point. Highly appreciated.

jneilliii, 2020-06-09:
So I just liked and for me the "classic studio" still had the same key. I wonder if it's only if you switch that it breaks the old way?

T3rmin8torX, 2020-06-10:
Very possible. Don't put any effort into it. I know what to do to make it
stream.

jneilliii, 2020-06-10:
Would it make it more convenient to add the steam key to the tab page? That wouldn't be too difficult to add.

T3rmin8torX, 2020-06-10:
It's not necessarily access to the stream controls. It's more of the
automatic start that it used to do when classic YouTube studio was the only
option.  You had it perfectly. Set the keys and everything worked. They
changed on their end. It's fine. I can adapt. 3d printing at it's best.

jneilliii, 2020-06-10:
Yeah, I know that's right. I just updated the plugin's readme with additional information related to this change. Appreciate the heads up.

closed 2020-06-10 by jneilliii