jneilliii/OctoPrint-YouTubeLive #61: TLS Issue

phpsystems, 2020-10-07:
From the docker commandline, I get:

https protocol not found, recompile FFmpeg with openssl, gnutls or securetransport enabled.

phpsystems, 2020-10-07:
More details:

docker run --device=/dev/vchiq --network="host" --name YouTubeLive -ti octoprint/youtubelive:latest https://192.168.42.40/webcam/?action=stream 4yr0-da6j-b3s7-vw9y-buse null
OctoPrint Webcam Stream: https://192.168.42.40/webcam/?action=stream
Live-stream secret: 4yr0-da6j-b3s7-vw9y-buse
Filter: null
ffmpeg version N-90990-g652b857b94 Copyright (c) 2000-2018 the FFmpeg developers
  built with gcc 6.3.0 (Raspbian 6.3.0-18+rpi1+deb9u1) 20170516
  configuration: --arch=armel --target-os=linux --enable-gpl --enable-omx --enable-omx-rpi --enable-nonfree
  libavutil      56. 18.100 / 56. 18.100
  libavcodec     58. 19.101 / 58. 19.101
  libavformat    58. 13.101 / 58. 13.101
  libavdevice    58.  4.100 / 58.  4.100
  libavfilter     7. 21.100 /  7. 21.100
  libswscale      5.  2.100 /  5.  2.100
  libswresample   3.  2.100 /  3.  2.100
  libpostproc    55.  2.100 / 55.  2.100
https protocol not found, recompile FFmpeg with openssl, gnutls or securetransport enabled.
https://192.168.42.40/webcam/?action=stream: Protocol not found
Did you mean file:https://192.168.42.40/webcam/?action=stream?

HAProxy is configured to do HSTS.

jneilliii, 2020-10-07:
This might be something broke on the upstream ffmpeg I'm referencing in my docker file. I assume you did all the stuff before that test command to download the github repo and build the docker image for youtubelive?

phpsystems, 2020-10-08:
I did, yes.

Just for clarity, I did:

cd ~
curl -sSL https://get.docker.com | sh
sudo usermod pi -aG docker
sudo reboot

cd ~
docker pull alexellis2/streaming:07-05-2018
git clone https://github.com/jneilliii/youtubelive --depth 1
cd youtubelive
docker build -t octoprint/youtubelive .

phpsystems, 2020-10-08:
I am looking in to allowing the webcam unencrypted, but that is more of a work around than a fix. I am also using a proper certificate (lets encrypt) rather than a self-signed one too.

jneilliii, 2020-10-08:
Relative to that, if you are using a default octopi image I believe you can use the non-ssl URL for the webcam stream, which of course would break it loading in the UI in some browsers now (ie chrome) and any port forwarding setups. Unfortunately, I'm not super familiar with creating a docker image for ffmpeg (the reason I'm using the other one) and will see if adding in that other stuff is too difficult or not. You may be able to work around this using the webcamstreamer plugin, as I believe it actually allows for a separate URL in it's settings for the source stream versus this one which is pulling from the default webcam & timelapse settings.

http://192.168.42.40:8080/?action=stream

jneilliii, 2020-10-08:
I'm looking into building my own ffmpeg now with ssl support and will report back in the next few days.

jneilliii, 2020-10-09:
Do me a favor please and test this command line via SSH and let me know if it actually attaches to your youtube stream or not. If it does then I think I can transition this plugin to utilize this other docker image as a base entry point. Make sure to replace the #### part with your stream secret.

```
docker run -it datarhei/ffmpeg:4.0.2-armhf -re -f mjpeg -framerate 5 -i https://192.168.42.40/webcam/?action=stream -ar 44100 -ac 2 -acodec pcm_s16le -f s16le -ac 2 -i /dev/zero -acodec aac -ab 128k -strict experimental -vcodec h264 -pix_fmt yuv420p -g 10 -vb 700k -framerate 5 -f flv rtmp://a.rtmp.youtube.com/live2/####-####-####-####-####
```

once it starts running you can press ctrl+c to close out the ffmpeg streaming.

jneilliii, 2020-10-09:
And one more command to test, just to see if I can eliminate docker from this plugin completely, reducing the complexity.

```
/usr/bin/ffmpeg -re -f mjpeg -framerate 5 -i http://192.168.0.120/webcam/?action=stream -ar 44100 -ac 2 -acodec pcm_s16le -f s16le -ac 2 -i /dev/zero -acodec aac -ab 128k -strict experimental -vcodec h264 -pix_fmt yuv420p -g 10 -vb 700k -framerate 5 -f flv rtmp://a.rtmp.youtube.com/live2/####-####-####-####-####
```

phpsystems, 2020-10-09:
------------------------------------------------------------------------------
OctoPrint version : 1.4.2
OctoPi version    : 0.17.0
------------------------------------------------------------------------------

pi@octoprint-01:~ $ docker run -it datarhei/ffmpeg:4.0.2-armhf -re -f mjpeg -framerate 5 -i https://192.168.42.40/webcam/?action=stream -ar 44100 -ac 2 -acodec pcm_s16le -f s16le -ac 2 -i /dev/zero -acodec aac -ab 128k -strict experimental -vcodec h264 -pix_fmt yuv420p -g 10 -vb 700k -framerate 5 -f flv rtmp://a.rtmp.youtube.com/live2/[Redacted]
Unable to find image 'datarhei/ffmpeg:4.0.2-armhf' locally
4.0.2-armhf: Pulling from datarhei/ffmpeg
95d54dd4bdad: Pull complete
72bf7d76c392: Pull complete
1e4608515d9b: Pull complete
ff34edff6cbb: Pull complete
817ab67755df: Pull complete
b0147ec1ea5d: Pull complete
158f2d6da6e4: Pull complete
a068b40c28c0: Pull complete
1422c8416a0d: Pull complete
3d0f97eb096e: Pull complete
9e791047d933: Pull complete
49f5d1051f8e: Pull complete
0bba6de42aee: Pull complete
0b27a2a4241b: Pull complete
163899275342: Pull complete
98f5b28c88c7: Pull complete
Digest: sha256:eaed294b85596c9b665a0856ae16b2b9ee77fdf9632c00571288ac86397f5082
Status: Downloaded newer image for datarhei/ffmpeg:4.0.2-armhf
ffmpeg version 4.0.2 Copyright (c) 2000-2018 the FFmpeg developers
  built with gcc 6.4.0 (Alpine 6.4.0)
  configuration: --bindir=/usr/local/bin --extra-libs=-ldl --extra-cflags=-I/usr/local/include --extra-ldflags=-L/usr/local/lib --extra-libs=-ldl --prefix=/usr/local --enable-nonfree --enable-gpl --enable-version3 --enable-avresample --enable-libmp3lame --enable-libx264 --enable-openssl --enable-postproc --enable-small --enable-libfdk_aac --enable-shared --disable-debug --disable-doc --disable-static
  libavutil      56. 14.100 / 56. 14.100
  libavcodec     58. 18.100 / 58. 18.100
  libavformat    58. 12.100 / 58. 12.100
  libavdevice    58.  3.100 / 58.  3.100
  libavfilter     7. 16.100 /  7. 16.100
  libavresample   4.  0.  0 /  4.  0.  0
  libswscale      5.  1.100 /  5.  1.100
  libswresample   3.  1.100 /  3.  1.100
  libpostproc    55.  1.100 / 55.  1.100
Input #0, mjpeg, from 'https://192.168.42.40/webcam/?action=stream':
  Duration: N/A, bitrate: N/A
    Stream #0:0: Video: mjpeg, yuvj420p(pc, bt470bg/unknown/unknown), 640x480, 5 tbr, 1200k tbn, 5 tbc
Guessed Channel Layout for Input Stream #1.0 : stereo
Input #1, s16le, from '/dev/zero':
  Duration: N/A, bitrate: 1411 kb/s
    Stream #1:0: Audio: pcm_s16le, 44100 Hz, stereo, s16, 1411 kb/s
Stream mapping:
  Stream #0:0 -> #0:0 (mjpeg (native) -> h264 (libx264))
  Stream #1:0 -> #0:1 (pcm_s16le (native) -> aac (native))
Press [q] to stop, [?] for help
[swscaler @ 0x23b1320] deprecated pixel format used, make sure you did set range correctly
[libx264 @ 0x234f610] using cpu capabilities: ARMv6 NEON
[libx264 @ 0x234f610] profile High, level 2.2
[libx264 @ 0x234f610] 264 - core 152 - H.264/MPEG-4 AVC codec - Copyleft 2003-2017 - http://www.videolan.org/x264.html - options: cabac=1 ref=3 deblock=1:0:0 analyse=0x3:0x113 me=hex subme=7 psy=1 psy_rd=1.00:0.00 mixed_ref=1 me_range=16 chroma_me=1 trellis=1 8x8dct=1 cqm=0 deadzone=21,11 fast_pskip=1 chroma_qp_offset=-2 threads=6 lookahead_threads=1 sliced_threads=0 nr=0 decimate=1 interlaced=0 bluray_compat=0 constrained_intra=0 bframes=3 b_pyramid=2 b_adapt=1 b_bias=0 direct=1 weightb=1 open_gop=0 weightp=2 keyint=10 keyint_min=1 scenecut=40 intra_refresh=0 rc_lookahead=10 rc=abr mbtree=1 bitrate=700 ratetol=1.0 qcomp=0.60 qpmin=0 qpmax=69 qpstep=4 ip_ratio=1.40 aq=1:1.00
Output #0, flv, to 'rtmp://a.rtmp.youtube.com/live2/g7fr-16ry-1v33-pzg2-0ygx':
  Metadata:
    encoder         : Lavf58.12.100
    Stream #0:0: Video: h264 (libx264) ([7][0][0][0] / 0x0007), yuv420p(progressive), 640x480, q=-1--1, 700 kb/s, 5 fps, 1k tbn, 5 tbc
    Metadata:
      encoder         : Lavc58.18.100 libx264
    Side data:
      cpb: bitrate max/min/avg: 0/0/700000 buffer size: 0 vbv_delay: -1
    Stream #0:1: Audio: aac ([10][0][0][0] / 0x000A), 44100 Hz, stereo, fltp, 128 kb/s
    Metadata:
      encoder         : Lavc58.18.100 aac
[mjpeg @ 0x22dfde0] Thread message queue blocking; consider raising the thread_queue_size option (current value: 8)
frame=   23 fps=5.0 q=14.0 size=      87kB time=00:00:00.02 bitrate=31023.3kbits/s speed=0.frame=   26 fps=5.1 q=12.0 size=     157kB time=00:00:00.62 bitrate=2056.8kbits/s speed=0.1frame=   28 fps=5.0 q=19.0 size=     203kB time=00:00:01.04 bitrate=1591.8kbits/s speed=0.1frame=   31 fps=5.1 q=19.0 size=     282kB time=00:00:01.62 bitrate=1420.5kbits/s speed=0.2frame=   33 fps=5.0 q=19.0 size=     309kB time=00:00:02.04 bitrate=1237.9kbits/s speed=0.3frame=   36 fps=5.1 q=14.0 size=     343kB time=00:00:02.62 bitrate=1071.1kbits/s speed=0.3frame=   39 fps=5.1 q=18.0 size=     365kB time=00:00:03.20 bitrate= 934.2kbits/s speed=0.4frame=   41 fps=5.1 q=17.0 size=     420kB time=00:00:03.64 bitrate= 944.9kbits/s speed=0.4frame=   44 fps=5.1 q=17.0 size=     451kB time=00:00:04.20 bitrate= 879.2kbits/s speed=0.4frame=   46 fps=5.1 q=14.0 size=     493kB time=00:00:04.64 bitrate= 869.0kbits/s speed=0.5frame=   49 fps=5.1 q=17.0 size=     520kB time=00:00:05.20 bitrate= 818.3kbits/s speed=0.5frame=   51 fps=5.0 q=17.0 size=     580kB time=00:00:05.64 bitrate= 842.6kbits/s speed=0.5frame=   54 fps=5.1 q=17.0 size=     615kB time=00:00:06.20 bitrate= 812.7kbits/s speed=0.5frame=   56 fps=5.0 q=14.0 size=     662kB time=00:00:06.64 bitrate= 816.2kbits/s  [... cut ...]
[libx264 @ 0x234f610] i8 v,h,dc,ddl,ddr,vr,hd,vl,hu:  6% 26% 53%  2%  2%  1%  2%  2%  4%
[libx264 @ 0x234f610] i4 v,h,dc,ddl,ddr,vr,hd,vl,hu: 21% 28% 24%  5%  5%  4%  4%  5%  5%
[libx264 @ 0x234f610] i8c dc,h,v,p: 52% 33% 13%  3%
[libx264 @ 0x234f610] Weighted P-Frames: Y:2.6% UV:0.0%
[libx264 @ 0x234f610] ref P L0: 56.1%  2.6% 24.5% 16.1%  0.7%
[libx264 @ 0x234f610] ref B L0: 70.9% 23.4%  5.7%
[libx264 @ 0x234f610] ref B L1: 83.3% 16.7%
[libx264 @ 0x234f610] kb/s:710.91
[aac @ 0x2350e30] Qavg: 65536.000
Exiting normally, received signal 2.

Appeared to work ok.

phpsystems, 2020-10-09:
The command line appeared to work as well. Rotating my octopi key, as I never seem to redact it properly ;-/

 /usr/bin/ffmpeg -re -f mjpeg -framerate 5 -i https://192.168.42.40/webcam/?action=stream -ar 44100 -ac 2 -acodec pcm_s16le -f s16le -ac 2 -i /dev/zero -acodec aac -ab 128k -strict experimental -vcodec h264 -pix_fmt yuv420p -g 10 -vb 700k -framerate 5 -f flv rtmp://a.rtmp.youtube.com/live2/g7fr-16ry-1v33-pzg2-0ygx
ffmpeg version 4.1.6-1~deb10u1+rpt1 Copyright (c) 2000-2020 the FFmpeg developers
  built with gcc 8 (Raspbian 8.3.0-6+rpi1)
  configuration: --prefix=/usr --extra-version='1~deb10u1+rpt1' --toolchain=hardened --incdir=/usr/include/arm-linux-gnueabihf --enable-gpl --disable-stripping --enable-avresample --disable-filter=resample --enable-avisynth --enable-gnutls --enable-ladspa --enable-libaom --enable-libass --enable-libbluray --enable-libbs2b --enable-libcaca --enable-libcdio --enable-libcodec2 --enable-libflite --enable-libfontconfig --enable-libfreetype --enable-libfribidi --enable-libgme --enable-libgsm --enable-libjack --enable-libmp3lame --enable-libmysofa --enable-libopenjpeg --enable-libopenmpt --enable-libopus --enable-libpulse --enable-librsvg --enable-librubberband --enable-libshine --enable-libsnappy --enable-libsoxr --enable-libspeex --enable-libssh --enable-libtheora --enable-libtwolame --enable-libvidstab --enable-libvorbis --enable-libvpx --enable-libwavpack --enable-libwebp --enable-libx265 --enable-libxml2 --enable-libxvid --enable-libzmq --enable-libzvbi --enable-lv2 --enable-omx --enable-openal --enable-opengl --enable-sdl2 --enable-omx-rpi --enable-mmal --enable-neon --enable-rpi --enable-libdc1394 --enable-libdrm --enable-libiec61883 --enable-chromaprint --enable-frei0r --enable-libx264 --enable-shared --libdir=/usr/lib/arm-linux-gnueabihf --cpu=arm1176jzf-s --arch=arm
  WARNING: library configuration mismatch
  avutil      configuration: --prefix=/usr --extra-version='1~deb10u1+rpt1' --toolchain=hardened --incdir=/usr/include/arm-linux-gnueabihf --enable-gpl --disable-stripping --enable-avresample --disable-filter=resample --enable-avisynth --enable-gnutls --enable-ladspa --enable-libaom --enable-libass --enable-libbluray --enable-libbs2b --enable-libcaca --enable-libcdio --enable-libcodec2 --enable-libflite --enable-libfontconfig --enable-libfreetype --enable-libfribidi --enable-libgme --enable-libgsm --enable-libjack --enable-libmp3lame --enable-libmysofa --enable-libopenjpeg --enable-libopenmpt --enable-libopus --enable-libpulse --enable-librsvg --enable-librubberband --enable-libshine --enable-libsnappy --enable-libsoxr --enable-libspeex --enable-libssh --enable-libtheora --enable-libtwolame --enable-libvidstab --enable-libvorbis --enable-libvpx --enable-libwavpack --enable-libwebp --enable-libx265 --enable-libxml2 --enable-libxvid --enable-libzmq --enable-libzvbi --enable-lv2 --enable-omx --enable-openal --enable-opengl --enable-sdl2 --enable-omx-rpi --enable-mmal --enable-neon --enable-rpi --enable-libdc1394 --enable-libdrm --enable-libiec61883 --enable-chromaprint --enable-frei0r --enable-libx264 --libdir=/usr/lib/arm-linux-gnueabihf/neon/vfp --cpu=cortex-a7 --arch=armv6t2 --disable-thumb --enable-shared --disable-doc --disable-programs
  avcodec     configuration: --prefix=/usr --extra-version='1~deb10u1+rpt1' --toolchain=hardened --incdir=/usr/include/arm-linux-gnueabihf --enable-gpl --disable-stripping --enable-avresample --disable-filter=resample --enable-avisynth --enable-gnutls --enable-ladspa --enable-libaom --enable-libass --enable-libbluray --enable-libbs2b --enable-libcaca --enable-libcdio --enable-libcodec2 --enable-libflite --enable-libfontconfig --enable-libfreetype --enable-libfribidi --enable-libgme --enable-libgsm --enable-libjack --enable-libmp3lame --enable-libmysofa --enable-libopenjpeg --enable-libopenmpt --enable-libopus --enable-libpulse --enable-librsvg --enable-librubberband --enable-libshine --enable-libsnappy --enable-libsoxr --enable-libspeex --enable-libssh --enable-libtheora --enable-libtwolame --enable-libvidstab --enable-libvorbis --enable-libvpx --enable-libwavpack --enable-libwebp --enable-libx265 --enable-libxml2 --enable-libxvid --enable-libzmq --enable-libzvbi --enable-lv2 --enable-omx --enable-openal --enable-opengl --enable-sdl2 --enable-omx-rpi --enable-mmal --enable-neon --enable-rpi --enable-libdc1394 --enable-libdrm --enable-libiec61883 --enable-chromaprint --enable-frei0r --enable-libx264 --libdir=/usr/lib/arm-linux-gnueabihf/neon/vfp --cpu=cortex-a7 --arch=armv6t2 --disable-thumb --enable-shared --disable-doc --disable-programs
  avformat    configuration: --prefix=/usr --extra-version='1~deb10u1+rpt1' --toolchain=hardened --incdir=/usr/include/arm-linux-gnueabihf --enable-gpl --disable-stripping --enable-avresample --disable-filter=resample --enable-avisynth --enable-gnutls --enable-ladspa --enable-libaom --enable-libass --enable-libbluray --enable-libbs2b --enable-libcaca --enable-libcdio --enable-libcodec2 --enable-libflite --enable-libfontconfig --enable-libfreetype --enable-libfribidi --enable-libgme --enable-libgsm --enable-libjack --enable-libmp3lame --enable-libmysofa --enable-libopenjpeg --enable-libopenmpt --enable-libopus --enable-libpulse --enable-librsvg --enable-librubberband --enable-libshine --enable-libsnappy --enable-libsoxr --enable-libspeex --enable-libssh --enable-libtheora --enable-libtwolame --enable-libvidstab --enable-libvorbis --enable-libvpx --enable-libwavpack --enable-libwebp - [... cut ...]
[libx264 @ 0x264d4f0] i8 v,h,dc,ddl,ddr,vr,hd,vl,hu:  6% 26% 53%  3%  2%  1%  2%  2%  4%
[libx264 @ 0x264d4f0] i4 v,h,dc,ddl,ddr,vr,hd,vl,hu: 19% 29% 25%  4%  5%  4%  4%  5%  5%
[libx264 @ 0x264d4f0] i8c dc,h,v,p: 51% 34% 13%  3%
[libx264 @ 0x264d4f0] Weighted P-Frames: Y:2.6% UV:0.0%
[libx264 @ 0x264d4f0] ref P L0: 55.9%  2.5% 24.4% 16.6%  0.7%
[libx264 @ 0x264d4f0] ref B L0: 70.4% 23.9%  5.6%
[libx264 @ 0x264d4f0] ref B L1: 84.2% 15.8%
[libx264 @ 0x264d4f0] kb/s:711.55
[aac @ 0x264ef30] Qavg: 65536.000
Exiting normally, received signal 2.

jneilliii, 2020-10-09:
Great, thanks for double-checking those for me. I think I'll go the route of converting the plugin to use ffmpeg directly and remove the docker dependency completely. This will take some time.