jneilliii/OctoPrint-YouTubeLive #62: Youtubelive plugin parameter

pierremilcent, 2020-10-08:
I installed the plugin and followed all configuration steps with success.

I was able to stream on youtube at the Test step.

I am blocked at the last one "OctoPrint Settings". 
[image]

What is this "YouTube Channel ID" ?

Thanks for your help

pierremilcent, 2020-10-08:
I found my channel ID in the advanced parameter section under Parameter in My Profile.
Here is what i get.

[image]

So now how to get the video stream from the camera ?

jneilliii, 2020-10-08:
Try following the instructions first. 

https://github.com/jneilliii/OctoPrint-YouTubeLive#requirements-for-streaming

jneilliii, 2020-10-09:
After looking at you screenshot again from my computer rather than my phone, it looks like you're getting the stream started fine? Are you able to access it from your YouTube Live dashboard to see the stream status?

pierremilcent, 2020-10-09:
Hello, 

no it does not work as expected. All steps with the test are OK, but when finishing and trying with the plugin it does not work. 
I am sot sure at all what are the channel id and stream id are. Here is what i used : 

And when i STOP Live here the message i get 

May i reinstall from scratch ? 

Regards. Cordialement. 

Pierre Milcent 

Viméo: https://www.vimeo.com/milcent 
Blog: https://milcent.wordpress.com 
Youtube: http://www.youtube.com/pmilcent 

----- Mail original -----

jneilliii, 2020-10-09:
If the test command worked, then the only thing that usually prevents the stream from working when you start it is changing the webcam and timelapse setting's stream URL to match the URL you used in the test command. The embedded viewer tab may be a problem related to embedding as I have been getting an error with mine for quite some time now and believe it to be some embed problem but haven't had the time to investigate further.

closed 2020-11-14 by stale[bot]