jneilliii/OctoPrint-YouTubeLive #64: Increase Frame Rate

AdamWP, 2021-01-10:
Is there any way to change the frame rate? The video is smooth in my octoprint control window, but choppy on my youtube live stream.

jneilliii, 2021-01-10:
If you edit the file entry.sh prior to compiling the docker part you can adjust the framerate. This is the line.

https://github.com/jneilliii/youtubelive/blob/f1f43da4579c9092551fc19da988257e4e9c1329/entry.sh#L7

AdamWP, 2021-02-06:
I was able to increase the framerate and it works for a while but then fails. Sometime's it's after 12 hours, sometimes 12 minutes. My internet connection is 500Mb/s and I'm running a raspberry pi 4 with 4GB of ram connected with an ethernet cable rather than wifi. htop reports my memory usage is below 300M and CPU load average is less than 30% Should this setup not be able to handle streaming 720p at 24fps?

It looks great in the octoprint control window, but streaming to youtube is very unreliable. I'd like to keep the resolution at least 720p or go much higher since I'm creating time lapses with octolapse. What kind of resolution, bit rates, and framerates are people streaming reliably?

jneilliii, 2021-02-06:
I'm not really sure, no one has ever reported any stats and since I've never used it long term don't have the personal details either. I would be curious if you get the same results just running the same ffmpeg command line directly via command line instead of inside the docker container. Might be a little easier to tweak all the various options with ffmpeg. The biggest drawback of the method being used in the plugin is re-encoding the stream rather than using some of the h264 video copy options using direct camera access like described where I got the original idea from [here](https://blog.alexellis.io/live-stream-with-docker/).

jneilliii, 2021-02-24:
I've made a little progress on decoupling the plugin from using docker. So as long as your device has ffmpeg (recent octopi versions have this be default), then you can try this new version. It gives full control of the ffmpeg commands with default options coming over from the existing docker entry.sh file. I haven't yet dealt with the filters (flip rotate options) but those can be added via command line too. This will allow you to adjust the framerate yourself. 

Install by copying/pasting the URL below into Plugin Manager > Get More > ...from URL.

```
https://github.com/jneilliii/OctoPrint-YouTubeLive/archive/develop/pure_ffmpeg.zip
```

I haven't tested this plugin fully, but very quick tests confirm that the plugin is able to stream to YouTube. The important part is that you have the ffmpeg command configured in OctoPrint's Timelapse Recording settings section.

[image]

jneilliii, 2021-05-11:
That's interesting, because that is exactly what the plugin is basically doing, running that command. Not sure what would cause it to just stop though.

AdamWP, 2021-05-13:
I'm getting the same behaviour even with "stream while printing" checked and printing directly from Octoprint rather than the SD card. It quits just after 4 minutes.

jneilliii, 2021-05-17:
I've updated the pure_ffmpeg branch with some additional hooks for logging, and a slightly different way of starting the command. I haven't yet figure out the best way to stop the stream once it's started yet, but at least should help with getting some logs. After install go into OctoPrint's logging section and set octoprint.plugins.youtubelive to debug and save the settings prior to starting the stream.

andyylin, 2022-05-30:
I'm also getting the issue of a stream stopping after just a few minutes (4-6 so far in my experience). In the YouTube Live tab, it still shows that it's live. Stopping and restarting with the button causes the live stream to start again just fine.

jneilliii, 2022-05-30:
Curious if you've tried my RTMPStreamer plugin yet to see if it does it? There was a lot of work that went into it by a contributor that removed docker requirement and uses ffmpeg command lines directly, giving gull control of all the settings. Also incorporated overlays but would try without them first because it adds additional processing time.

AdamWP, 2022-05-30:
Hi. I just installed it directly in Octoprint on a couple machines but I get the following error pop up in the top right corner of Octoprint:

YouTube Live Error
('Connection aborted.', FileNotFoundError(2, 'No such file or directory'))

jneilliii, 2022-05-30:
That error would exist if you were running the docker version of this plugin but don't have the docker bits installed I think?

AdamWP, 2022-05-30:
That was it. I had installed v0.5.2 through the plugin manager rather than running https://github.com/jneilliii/OctoPrint-YouTubeLive/archive/develop/pure_ffmpeg.zip.

Now I have 0.6.0rc2 installed and it's working(It does ask to upgrade to 0.5.0 btw).  No more having to ssh into each machine now. I'll try running a few long prints and see how it works. 

Great update!

andyylin, 2022-05-31:
> Curious if you've tried my RTMPStreamer plugin yet to see if it does it? There was a lot of work that went into it by a contributor that removed docker requirement and uses ffmpeg command lines [...]

Thanks for the suggestion, I just switched to RTMPStreamer and got it working on Twitch. Running the stream gets my Raspberry Pi 3B+ at around 50-60% CPU though. Do you recommend upgrading to a Raspberry 4 for live streaming?

jneilliii, 2022-05-31:
It would help but would still maintain high CPU load because of transcoding the video from one format (mjpg) to another (h264 flv). You could try tweaking the options for the ffmpeg command, but did find Twitch to be a little more reliable than YouTube Live for some reason. In all cases ethernet connection is ideal for live streaming over WiFi. 

A third option might be using restreamer over either plugin and have it supply HLS for OctoPrint, and can stream direct to external source (Twitch). I have notes on setting that up [here](https://gist.github.com/jneilliii/fb40f42d16887649b58699f002fe48ec).

andyylin, 2022-06-02:
> A third option might be using restreamer over either plugin and have it supply HLS for OctoPrint, and can stream direct to external source (Twitch). I have notes on setting that up [here](https://gist.github.com/jneilliii/fb40f42d16887649b58699f002fe48ec).

Thanks, I'm giving it a try, but running into a question that I've posted as a comment on that note.

jneilliii, 2022-06-02:
just in case you don't get replies to gist...

/dev/video0 looks like a raspicam style camera connection on the CSI connection and in most cases will be the correct device if there is only one camera.

Puntoboy, 2022-07-28:
I'm getting a similar issue. Just installed this with the latest version of OctoPi. Choppy video and errors from Youtube.

[image]

jneilliii, 2022-07-28:
Welcome to the club. It's been that way since this plugin's inception. If you have any ideas I'm all ears, although I've considered just dropping this plugin completely in favor of my other more generic rtmpstreamer plugin that doesn't require docker.