jneilliii/OctoPrint-YouTubeLive #65: 409 Client Error: Conflict ("Conflict. The container name "/YouTubeLive"

samkinsey2, 2021-01-29:
This is the response that i get when i try to start the stream. i am hoping that it something that has come up before and it is an easy fix.
octoprint version Version 1.3.12

  409 Client Error: Conflict ("Conflict. The container name "/YouTubeLive" is already in use by container "13e0420b80626ff222f9efc7de51794a6884cc1e63dd71e6cbf4ecdc87b2d619". You have to remove (or rename) that container to be able to reuse that name.")

jneilliii, 2021-01-29:
Looks like you didn't stop and remove the container from the initial testing commands?

samkinsey2, 2021-01-29:
Yes i figured it out but is there a way to embed this in the plugin to
avoid opening putty and  (docker rm -f YouTubeLive) everytime you are done.

jneilliii, 2021-01-29:
It shouldn't be happening if you are using the stop button to end the stream. If it is then there is definitely a bug or something changed in docker possibly. I don't know that side of things very well.

samkinsey2, 2021-01-29:
Ok that's what I was thinking, I dont know that much about docker either. I
guess it is what it is until some one can figure it out. It won't shut off
until I started using the -f command to force rm container. Then I have to
physically enter the docker run ect ect ect....to start a new container and
run it again..
Thank you for the response, I usually don't have to ask for help like this
so this is my first time asking ever in a form so thank you...🙂

Sam Kinsey

jneilliii, 2021-01-29:
You shouldn't have to do the run command, that's what the go live button will do for you as well. After initial testing all control should be done from the button.

samkinsey2, 2021-01-29:
It doesn't work like that for me I have to rm -f the container then if  I
try to start the stream in octoprint it says no container found youtube
ect. or somthing like that I am running a print now so I can not give you
the exact response. so I go back in to putty and use the run command . Go
back in to octoprint then hit start it works. It is a pain . Unless I am
missing somthing simple and am just an idiot. 😆
It would not be the first time and definitely not the last..

Sam Kinsey

jneilliii, 2021-01-30:
So I went through the instructions on a recently flashed octopi 0.18 image and am unable to reproduce your issue. I made sure the container was stopped and removed prior to closing my SSH session with the test command and the go live button and stop button are working as expected so I'm not quite sure what you may be running into here. The one issue I do see is that the tab doesn't appear to be picking up my live channel when I play, as it thinks I'm offline, but looking at the Live Stream dashboard it is definitely broadcasting, so I assume that is something to do with the new live dashboard maybe, that I'm unsure of.

samkinsey2, 2021-01-30:
Maybe I just need to upgrade octoprint. I am running 2 instance and last
time I upgraded it was a disaster.

Sam Kinsey

closed 2021-02-21 by github-actions[bot]