jneilliii/OctoPrint-YouTubeLive #66: Docker Headaches

jneilliii, 2021-03-04:
Quoted from https://github.com/jneilliii/OctoPrint-RTMPStreamer/issues/28#issue-821877228 by @epn35.

Hello,
I have been confused for a few days with Docker. I'm trying to stream with "RTMP Streamer" or "WebcamStreamer" but it's the same problem: `<404 Client Error: Not Found ("pull access denied for adilinden / rpi-ffmpeg, repository does not exist or may require 'docker login ': denied: requested access to the resource is denied ")>`

[image]
[image]

Have you ever had this problem? Can you guide me to a solution, I don't know what to test anymore ..
I don't have proxy or something and i have corect link in webcam & Timelaps :/
Thank you.

jneilliii, 2021-03-04:
I think the issue that you're probably running into is some form of network routing/access issue from within docker. If I understand correctly, your OctoPrint instance is also in docker? Are you running the youtubelive container within the octoprint docker container or not?

epn35, 2021-03-04:
Hello, thanks for moving the post.
I'm sorry, I don't understand everything, I installed octoprint on a raspberry, then then the youtubelive plugin through the interface.
I can't understand the function of Docker, it's complicated to give you informations and find solutions: /
I put a DMZ on the raspberry ip to see if there was a routing / access problem, but it doesn't change anything.
What can I do to give you information that can help you?

jneilliii, 2021-03-04:
Ok, let me verify a couple of things. You installed OctoPrint using the octopi image and you're just using the docker stuff for the YouTubeLive integration? If that is the case then you shouldn't be having any issues outside of the tab not being able to show the live stream (YouTube changed their embedding requirements). The stream should still broadcast to YouTube as long as it can reach the internet. Looking at your screenshot again it appears that it is sending frames. Did you check the stream from the YouTube Live Studio?

jneilliii, 2021-03-04:
Also, looking at the webcamstreamer plugin, adilinden appears to have removed his docker image from docker hub that is pi compatible in favor of a generic x86 or x86_64 Docker host.

jneilliii, 2021-03-04:
if you want to try the webcamstreamer one you can reference this issue for getting the docker image for it.

https://github.com/adilinden-oss/octoprint-webcamstreamer/issues/39#issuecomment-753514737

closed 2021-03-27 by github-actions[bot]