jneilliii/OctoPrint-YouTubeLive #67: Connection error?

FavSOB, 2021-03-13:
Still new to this. Appreciate any assistance. 

I have followed your instructions all the way down to the TEST command. When I input the string:
`docker run --device=/dev/vchiq --network="host".....`

I receive several lines of text, followed by this error.  
Connection to tcp://192.168.1.60:8080 failed: Connection refused http://192.168.1.60:8080/?action=stream: Connection refused

[image]

I have not had any previous issues connecting to Octoprint, and have been using it for over a week before deciding to try the YouTubeLive plugin.
Not sure the reason behind this message. 

[image]

Any thoughts would be appreciated.

jneilliii, 2021-03-13:
I think if you're using a newer octopi 0.18 image this specific test URL may be broken Try with this test URL and see if it works as expected.

```
docker run --device=/dev/vchiq --network="host" --name YouTubeLive -ti octoprint/youtubelive:latest  http://192.168.1.60/webcam/?action=stream xxxx-xxxx-xxxx-xxxx null

```

FavSOB, 2021-03-13:
Ok!  now I am getting past that error!  thanks!   
The connection problem has been addressed. Thanks!

But now I get all of this....

OctoPrint Webcam Stream: http://192.168.1.60/?action=stream
Live-stream secret: xxxxxxxxxxxxxxx
Filter: null
ffmpeg version N-90990-g652b857b94 Copyright (c) 2000-2018 the FFmpeg developers
  built with gcc 6.3.0 (Raspbian 6.3.0-18+rpi1+deb9u1) 20170516
  configuration: --arch=armel --target-os=linux --enable-gpl --enable-omx --enab    le-omx-rpi --enable-nonfree
  libavutil      56. 18.100 / 56. 18.100
  libavcodec     58. 19.101 / 58. 19.101
  libavformat    58. 13.101 / 58. 13.101
  libavdevice    58.  4.100 / 58.  4.100
  libavfilter     7. 21.100 /  7. 21.100
  libswscale      5.  2.100 /  5.  2.100
  libswresample   3.  2.100 /  3.  2.100
  libpostproc    55.  2.100 / 55.  2.100
[mjpeg @ 0x2d4f2f0] No JPEG data found in image
[mjpeg @ 0x2d4c490] decoding for stream 0 failed
[mjpeg @ 0x2d4c490] Could not find codec parameters for stream 0 (Video: mjpeg,     none(bt470bg/unknown/unknown)): unspecified size
Consider increasing the value for the 'analyzeduration' and 'probesize' options
Input #0, mjpeg, from 'http://192.168.1.60/?action=stream':
  Duration: N/A, bitrate: N/A
    Stream #0:0: Video: mjpeg, none(bt470bg/unknown/unknown), 5 tbr, 1200k tbn,     5 tbc
Guessed Channel Layout for Input Stream #1.0 : stereo

jneilliii, 2021-03-13:
Yeah, go back to the instructions from that point, to cancel out and remove the docker container you added with that test command, etc.

FavSOB, 2021-03-13:
Still got that "no JPEG data foud in image" error.. but went ahead and started a print job, and guess what?
WORKS LIKE A CHARM!   Thanks for working on this and thanks for your support!
Can't thank you enough :)
Scott

closed 2021-03-13 by FavSOB

jneilliii, 2021-03-13:
Thank you for the tip, glad it worked out. Unfortunately, the tab probably won't work for viewing the live stream due to recent changes by Google, where the embed has to include a public url in the stream parameter.

FavSOB, 2021-03-14:
Oh, you know what? I haven’t even looked at the tab inside octoprint to see if it does or not.  I actually just watch it via the CONTROL tab anyways, so if it doesn’t that’s fine.

 

Based on some of the information I was curious if you are continuing to support and develop this plugin?

Thanks!

Scott

jneilliii, 2021-03-14:
I've been working on changes here and there. I was working on a way to remove the docker dependency, but haven't got that far yet with re-programming all that.