jneilliii/OctoPrint-YouTubeLive #68: docker error

Z0mb1ee, 2021-04-28:
when i run:
`docker run --device=/dev/vchiq --network="host" --name YouTubeLive -ti octoprint/youtubelive:latest http://<ip>:8080/?action=stream xxxx-xxxx-xxxx-xxxx null`
i get:
`docker: Error response from daemon: error gathering device information while adding custom device "/dev/vchiq": no such file or directory.
ERRO[0000] error waiting for container: context canceled `
where is the problem?

jneilliii, 2021-04-28:
`--device=/dev/vchiq` is the parameter that grants access to the underlying GPU of the pi. The error indicates that the device doesn't exist, so try the command without that parameter and see if it works.

Z0mb1ee, 2021-04-29:
`sudo docker run --network="host" --name YouTubeLive -ti octoprint/youtubelive:latest http://192.168.31.7:8080/?action=stream `

`stfs-gvch-gzdf-uj18-4w6b null
OctoPrint Webcam Stream: http://192.168.31.7:8080/?action=stream
Live-stream secret: stfs-gvch-gzdf-uj18-4w6b
Filter: null
ffmpeg version N-90990-g652b857b94 Copyright (c) 2000-2018 the FFmpeg developers
  built with gcc 6.3.0 (Raspbian 6.3.0-18+rpi1+deb9u1) 20170516
  configuration: --arch=armel --target-os=linux --enable-gpl --enable-omx --enable-omx-rpi --enable-nonfree
  libavutil      56. 18.100 / 56. 18.100
  libavcodec     58. 19.101 / 58. 19.101
  libavformat    58. 13.101 / 58. 13.101
  libavdevice    58.  4.100 / 58.  4.100
  libavfilter     7. 21.100 /  7. 21.100
  libswscale      5.  2.100 /  5.  2.100
  libswresample   3.  2.100 /  3.  2.100
  libpostproc    55.  2.100 / 55.  2.100
Input #0, mjpeg, from 'http://192.168.31.7:8080/?action=stream':
  Duration: N/A, bitrate: N/A
    Stream #0:0: Video: mjpeg, yuvj420p(pc, bt470bg/unknown/unknown), 640x480 [SAR 1:1 DAR 4:3], 5 tbr, 1200k tbn, 5 tbc
Guessed Channel Layout for Input Stream #1.0 : stereo
Input #1, s16le, from '/dev/zero':
  Duration: N/A, bitrate: 1411 kb/s
    Stream #1:0: Audio: pcm_s16le, 44100 Hz, stereo, s16, 1411 kb/s
Stream mapping:
  Stream #0:0 -> #0:0 (mjpeg (native) -> h264 (h264_omx))
  Stream #1:0 -> #0:1 (pcm_s16le (native) -> aac (native))
Press [q] to stop, [?] for help
[swscaler @ 0x1fb85a0] deprecated pixel format used, make sure you did set range correctly
* failed to open vchiq instance`

Z0mb1ee, 2021-04-29:
is there a problem because my octoprint does not run on raspberry pi?
i have it on orange pi one.

jneilliii, 2021-04-29:
Good point. The ffmpeg command that is being run inside the docker container is actually utilizing the OMX GPU Driver available on the Raspberry Pi. If that doesn't work the same on the orange then this specific ffmpeg image will not work for you. I've been working in my spare time to remove the docker requirement and just run the ffmpeg command directly, but development has been slow.

Z0mb1ee, 2021-04-29:
so it will not work and i can uninstall it?

jneilliii, 2021-04-29:
yeah, it probably won't work for you until those other changes are completed. You could look into the webcam streamer plugin, but there are some of it's instructions that have become deprecated, but it allows you to use whatever specific ffmpeg docker image you can find to run it.

Z0mb1ee, 2021-04-29:
ok

closed 2021-04-29 by Z0mb1ee