jneilliii/OctoPrint-YouTubeLive #77: OctoPrint-YouTubeLive stream orientation and mirror

Kolokial, 2024-04-25:
I cannot see how to stream the orientation and mirrored effects that OctoPrint has applied. Is this a new feature, or something I am missing?

jneilliii, 2024-04-25:
This may be related to #71 and not retrieving the correct settings after they've been moved into a plugin setting instead of a global setting. Assuming you're on OctoPrint 1.9.0+ are you using the classic webcam settings plugin, or another webcam provider plugin for embedding the stream (ie multicam or camera-streamer control)?

Kolokial, 2024-04-26:
Thanks for the response.

I'm using the classic webcam.

I'm not sure what you are referring to with settings being moved?

jneilliii, 2024-04-27:
try the new version 0.5.3 that I just released, it should pick up those settings again now. There was a change in OctoPrint 1.9.0 where those webcam settings went from global settings into a plugin setting. This plugin wasn't upgraded to handle that change, and therefore the rotation and flip options were not getting picked up. They should be now with this new version.

closed 2024-05-20 by github-actions[bot]